stm32-rs/stm32f1xx-hal #2: Cleanup examples

TeXitoi, 2018-12-13:
Maybe we can cleanup examples that doesn't show how to use this crate or test one of this crate functionnality? I think in particular to remove the examples relying on git dependencies.

I don't know if an example using RTFM (once 0.4 is out) is relevant. What do you think?

therealprof, 2018-12-13:
I think we should move out the examples entirely into a stm32f103-examples (sub-)crate or some potentially into their own crates. Anything that lives in the crate itself should would need to work on any MCU in that family (and since that requires a memory.x that would be kind of meh).

TeXitoi, 2018-12-13:
Having example can be really interesting. For example, to check that binaries still works on PR, of on PR to give something to test on real hardware.

I don't know if we can have more than one configuration on examples, to be able to compile for more than one MCU.

therealprof, 2018-12-13:
Oh, I agree that having code exercising functionality to test for proper compilation would be a nice thing. But the examples here are highly target dependent and need additional hardware to be runnable at all so I'd prefer to have it in board specific repos rather than in the multi-MCU HAL if that makes sense.

TeXitoi, 2018-12-13:
adding an stm32f103-examples, and build them on std32f1xx's travis?

We'll have to find a way to use the current code with the examples to do that, but it looks like a good solution.

therealprof, 2018-12-13:
> adding an stm32f103-examples, and build them on std32f1xx's travis?

Yes, that.

therealprof, 2018-12-13:
Or move it out if it's really more a project in its own.

TheZoq2, 2019-02-10:
A crate in the repo sounds like a better idea to me, if nothing else it makes the examples easier to find

burrbull, 2019-02-11:
An idea.
Add `cargo generate` option `--from-example examplename`. This will move example into main.rs.

We can store all templates and examples for different microcontrollers in the one repository in different branches (already working) or in different folders.

therealprof, 2019-02-11:
@burrbull Sorry, but now I'm confused. :-D Which crate is the one to rule them all and who's going to maintain that?

burrbull, 2019-02-11:
```
cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart --branch stm32f103c8 --from-example blinky
cargo run
```

TheZoq2, 2020-10-05:
I think the current example setup works quite well, so 'll close this for now

closed 2020-10-05 by TheZoq2