stm32-rs/stm32f1xx-hal #6: GPIO AF should be activated after enabling the corresponding peripheral

WGH-, 2019-01-05:
According to the manual ("9.1.4 Alternate functions (AF)"),
> If software configures a GPIO pin as Alternate Function Output, but peripheral is not
activated, its output is not specified.

So it will be cleaner to enable the AF only after enabling the corresponding peripheral.

Not sure if it was caused by this, but I certainly saw something that looked like spurious SPI clock signals.

I'm not sure how to fix the problem cleanly. Perhaps, the pin struct should have methods `activate_af` and `deactivate_af`,`so the peripheral constructor will activate AF after enabling peripheral itself, and deactivate AF before disabling the peripheral (disabling peripherals doesn't seem to be implemented for any peripheral at the moment, though).

Windfisch, 2019-10-22:
Is this an issue for the other STM32Fx series as well?

burrbull, 2019-10-22:
What about move all `into_alternate..` in `IntoAF` trait and add `where IntoAF` for all PINS in peripheral constructors?

TheZoq2, 2019-10-23:
Hmm, interesting issue, and probably something that should be fixed. I feel like the easiest fix for it would be to disable/hide `into_alternate...`, and make the peripheral constructors themselves call it.

> I'm not sure how to fix the problem cleanly. Perhaps, the pin struct should have methods activate_af and deactivate_af,`so the peripheral constructor will activate AF after enabling peripheral [...]

I feel like we can just return them in any suitable mode that isn't AFIO. Floating input would probably be the best option

> Is this an issue for the other STM32Fx series as well?

My guess is yes, but @therealprof probably knows more.

burrbull, 2019-12-07:
I tried to use `FnOnce` closure:

https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=637cec5d02765c4d5c8a11cda8dd66c5