stm32-rs/stm32f1xx-hal #7: Calulation of  brr for Serail Devices with Clocksource PCLK1

knoby, 2019-01-06:
The Calculation of the BRR Register is not correct if the clock is not pclk2. In [serial.rs:130](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/serial.rs) the brr value is calculated with the clock rate from pclk2 even if the device is using pclk1:

`let brr = clocks.pclk2().0 / baud_rate.0;`

pclk2 is only used for USART1, the others use pclk1. 
Quote from the reference manual p. 796 27.3.4 "Fractional baud rate generation":
> Input clock to the peripheral (PCLK1 for USART2, 3, 4, 5 or PCLK2 for USART1)

If pclk1 and pclk2 have different clock frequencies the baud rate of the USART device is not as expected:

This would NOT work for USART2, 3, 4 and 5 but for USART1
`let clocks = rcc.cfgr.sysclk(64.mhz()).pclk1(32.mhz()).pclk2(64.mhz()).freeze(&mut flash.acr);`

This would work for USART2, 3, 4 and 5 but for USART1
`let clocks = rcc.cfgr.sysclk(64.mhz()).pclk1(32.mhz()).pclk2(32.mhz()).freeze(&mut flash.acr);`

Depending on the USART device the correct clock frequency has to be selected to calculate the brr register.

gbip, 2019-02-19:
Since I had the same problem today, I will take a look at this and make a PR in the following days.

nraynaud, 2019-03-10:
can we close this?

TheZoq2, 2019-03-10:
Looks like it, thanks for the reminder!

closed 2019-03-10 by TheZoq2