stm32-rs/stm32f1xx-hal #8: The documentation link on crates.io doesn't work

Mendelt, 2019-01-07:
It looks like the documentation link in Cargo.toml is broken. It is "https://docs.rs/crate/stm32f1xx-hal" but I think it should be "https://docs.rs/stm32f1xx-hal" without crate.
Or you can just leave the documentation key out. Crates.io auto-generates a link to docs.rs if none is specified.
I'm not sure if this is the only problem though because the documentation doesn'tseem to have been generated on docs.rs at all for this crate.

TheZoq2, 2019-02-10:
Looks like the link was fixed in 67aa99e but the docs generation is still broken, most likely because the required changes to Cargo.toml (`[package.metadata.docs.rs]`) have not been published to docs.rs yet. @therealprof should we push a new version or is there something else that needs fixing before we do that?

Mendelt, 2019-02-10:
No I think you're right. I already saw the fix but was waiting for the change to be published. I think docs.rs pulls from crates.io so the docs should show up after publishing.

therealprof, 2019-02-10:
@TheZoq2 Yeah, we should also bump the `stm32f1` version.

closed 2019-02-10 by Mendelt