stm32-rs/stm32f1xx-hal #11: Document the features = [stm32f103] requirement

TheZoq2, 2019-02-09:
Given that the old stm32f103_hal crate is now deprecated, I figured I should update my projects. The first hurdle I ran into was  a couple of `unresolved import stm32` errors as well as `duplicate definitions with name <name>` errors. I managed to fix this once I figured out that you need to add a feature to specify which device you are compiling for.

However, this certainly wasn't obvious so it would be nice if it was added to the readme.

therealprof, 2019-02-09:
Absolutely! Please feel free to send a PR. I'm also looking for people who're willing to share maintainership of this crate...

TheZoq2, 2019-02-09:
> Absolutely! Please feel free to send a PR.
Done, #12 

> I'm also looking for people who're willing to share maintainership of this crate...

I wouldn't mind helping maintain this :)

closed 2019-02-09 by TheZoq2