stm32-rs/stm32f1xx-hal #13: ITM logging not working in windows due to auto-load safe-path

ninjasource, 2019-02-09:
Hi 

Has anyone tried running this from Windows 10? I get the following warning message when running the following command:
`cargo run --example itm --features="stm32f103,rt"`

The gdb warning below assumes a linux or osx operating system and I was wondering if ITM logging works at all in Windows. 

I tried to create a file `C:\Users\xxx\.gdbinit` with `set auto-load safe-path /` in contents but I get the same warning. 

`
warning: File "C:\Users\xxx\Source\Repos\stm32f1xx-hal\.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
        add-auto-load-safe-path C:\Users\xxx\Source\Repos\stm32f1xx-hal\.gdbinit
line to your configuration file "$HOME/.gdbinit".
To completely disable this security protection add
        set auto-load safe-path /
line to your configuration file "$HOME/.gdbinit".`

ninjasource, 2019-02-09:
Also tried `set auto-load safe-path \` in case it was a path character issue

TheZoq2, 2019-02-10:
I don't have a windows pc to test this on, but my suspicion is that `$HOME` isn't set to your user directory which is suggested by this SO answer https://stackoverflow.com/a/52066014

TheZoq2, 2019-10-07:
@ninjasource Did you find a solution to this problem or is it still relevant?

ninjasource, 2019-10-07:
Hi, thanks for resurrecting this thread! I have published a Windows command line tool that does what you can easily do in Linux: Print ITM packets from the com port to the console

serialitm = "0.1.0"

https://github.com/ninjasource/serialitm

David

closed 2019-10-09 by ninjasource

burrbull, 2019-10-09:
> https://github.com/ninjasource/serialitm

It's make sense to add this to [awesome-embedded-rust](https://github.com/rust-embedded/awesome-embedded-rust)

ninjasource, 2019-10-09:
Thanks, I'll do that!