stm32-rs/stm32f1xx-hal #16: Exporting the sdv2rust crate as device

TeXitoi, 2019-02-11:
For consistency, I propose to re export the sdv2rust crate as `device`. It is coherent with the name used by RTFM: https://japaric.github.io/cortex-m-rtfm/book/by-example/app.html

If there is approvals and no veto, I can propose such a PR.

therealprof, 2019-02-11:
Actually the agreed on terminology is `pac`. I would not go as far as insta-dropping `stm32` but we can certainly mark it is deprecated and re-export the PAC as `pac`.

therealprof, 2019-02-11:
Funny enough even the proponents of that terminology are not using `pac` to re-export the PAC:
```pub use nrf52832_pac as target;```

TheZoq2, 2019-02-11:
Changing the name does sound like a good idea, I don't really like the current `stm32` name, `device` makes a lot more sense. However, if `pac` is the agreed upon terminology, we should probably do that.

Out of curiosity, why was `pac` chosen, I don't really understand what it would mean in this context.

therealprof, 2019-02-11:
`pac` := Peripheral Access Crate

therealprof, 2019-02-11:
cf. https://rust-embedded.github.io/book/portability/index.html

TeXitoi, 2019-02-11:
Of course we will do that with a new export. I don't really care of the name as long as it can be common between all hal implementation (not only stm32).

The ambiguity with `pac` is that there are 2 PAC crates: the core PAC (cortex-m) and the device PAC (stm32f1 in this case).

therealprof, 2019-02-11:
Is that so? I haven't thought of `cortex-m` as a PAC so far. Maybe we should raise the question as issue on the embedded-wg issue tracker to sort out potential conflicts before settling on an implementation.

TheZoq2, 2019-02-11:
If nothing else, the figure in the embedded book seems to indicate that the PAC is the only layer between the microcontroller and the HAL. A wg issue sounds like a good idea to get clarifications about this

therealprof, 2019-02-11:
Well, don't let the figure irritate you; I've created that. ;)

However https://rust-embedded.github.io/book/start/registers.html agrees with that view and that hadn't been done by me.

TeXitoi, 2019-02-11:
The thing is that makes me saying that cortex-m is a "PAC" is that there is a `Peripheral` object in cortex-m: https://docs.rs/cortex-m/0.5.8/cortex_m/peripheral/struct.Peripherals.html

TeXitoi, 2019-02-11:
In RTFM, there is a core peripheral object and a device peripheral object that is given to the init function.

TheZoq2, 2019-02-22:
#29 Closed this automatically but I don't think it's fixed since we didn't agree upon a name

therealprof, 2019-02-22:
@TheZoq2 Uhm, but with the latest changes we're exporting it as `stm32`, `pac` and `device`.

TheZoq2, 2019-02-22:
Yea, that's true, but I feel like we should eventually decide on one of them to avoid confusion. Allthough I suppose allowing all three also has some benefits

therealprof, 2019-02-22:
Well, if we have to chose one it should probably the documented one which is `pac`. 😏

PR #29 (2019-02-22): Edition 2018
2018 edition path style.

Closes #16 .