stm32-rs/stm32f1xx-hal #18: Feature: fmt::Write for serial::Tx

jounathaen, 2019-02-14:
Hi,
I'm new to this HAL, rather new to Rust and I still have a lot of confusion, so apologize if this does not belong here or is a stupid question. 

I'm using the usart (like in the example) which provides the function `Tx.write(byte)`. But there is no functionality to send a (formatted) string provided.

This can be implemented with something like:
```
use core::fmt::{self, Write};

struct Format_Tx {
    Tx :  stm32f1xx_hal::serial::Tx<USART1>,
}

impl fmt::Write for Format_Tx {
    fn write_str(&mut self, s: &str) -> fmt::Result {
        for byte in s.bytes() {
            block!(self.Tx.write(byte)).ok();
        }
        Ok(())
    }
}
```

This enables the write macro, if you create a `Format_Tx` "relay" struct. For example `write!(my_format_tx, "Formatted {} String", 123);`

Could  `stm32f1xx_hal::serial::Tx`  implement this?
Then writing strings to the serial out would be as easy as `write!(mytx, "foobar")`. I guess this is a common use case.

Regards Jounathaen

therealprof, 2019-02-14:
@jounathaen Indeed we should have such an implementation. It's probably just an oversight that it doesn't exist yet and there's no blanket implementation.

It shouldn't be necessary to have your proxy structure, you can impl it directly on the Tx struct, something like (copied from stm32f0xx-hal and untested on stm32f1xx-hal):
```
impl<USART> Write for Tx<USART>
where
    Tx<USART>: embedded_hal::serial::Write<u8>,
{
    fn write_str(&mut self, s: &str) -> Result {
        s.as_bytes()
            .iter()
            .try_for_each(|c| nb::block!(self.write(*c)))
            .map_err(|_| core::fmt::Error)
    }
}
```

TeXitoi, 2019-02-14:
Shouldn't this impl be directly in embedded_hal?

therealprof, 2019-02-14:
@TeXitoi I think it should be possible to have a blanket impl in embedded_hal.

TeXitoi, 2019-02-14:
```rust
impl<T> Write for T where T: embedded_hal::serial::Write<u8>
```

There is no embedded_hal type, thus I think the orphan rule prohibit it (not sure).

PR #49 (2019-03-19): feat: fmt::Write for serial Tx
Closes #18, all credit to @therealprof.
closed 2019-03-19 by therealprof