stm32-rs/stm32f1xx-hal #24: I2c Example

andrewmruddy, 2019-02-17:
*New to Rust* - Would it be possible to have an example for I2c? After looking through the documentation, The Rust Book, and other examples, I am still having trouble figuring out proper implementation.

TheZoq2, 2019-02-18:
Writing an example without using a specific device is difficult.

I will try to write some documentation for the i2c device though, is there anything in particular you're having trouble with?

andrewmruddy, 2019-02-18:
I have the following for setting up the i2c peripheral: 
`
    let peripherals = stm32f103::Peripherals::take().unwrap();
    let mut gpiob = peripherals.GPIOB.split(&mut rcc.apb2);
    let sda = gpiob.pb7.into_alternate_open_drain(&mut gpiob.crl);
    let scl = gpiob.pb6.into_alternate_open_drain(&mut gpiob.crl);

    let i2c_bus = I2c::i2c1(
        peripherals.I2C1,
        (scl,sda),
        &mut afio.mapr,
        hal::i2c::I2c::Mode::Standard, // < i am having problems with the enum "Mode"
        clocks,
        &mut rcc.apb1,
    );` I am using the MMA8451 accelerometer (adafruit breakout board) for my proof-of-concept.

therealprof, 2019-02-18:
@andrewmruddy There're some examples for I2C use with `stm32f1xx-hal` in the `ssd1306` crate, e.g. https://github.com/jamwaffles/ssd1306/blob/master/examples/graphics_i2c.rs

andrewmruddy, 2019-02-18:
Thank you so much!

jounathaen, 2019-02-19:
If you have a working example, you could post it here.
This would surely help others (like me, I'm planning to do something simmilar :-D )

therealprof, 2019-02-19:
@jounathaen I wouldn't want to add external dependencies for examples and unfortunately I2C without connected devices is not too impressive/useful.

jounathaen, 2019-02-19:
@therealprof I can understand that dependencies are undesired, but first I think that you can have I2C examples without external dependencies, like reading generic registers, and second I think that having an outdated example is still better than no up-to-date example.
I was not necessarily thinking about full blown working examples, but code snippets for the documentation would be helpful.
(Of course just my humble opinion)

therealprof, 2019-02-19:
@jounathaen Oh sure, I definitely agree we should have proper code documentation with logical code snippets.

jaxter184, 2021-01-13:
The [i2c scanner example in the stm32f3xx-hal](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/examples/i2c_scanner.rs) may be a good reference for including an i2c example that doesn't depend on any connected devices

JoaquinEduardoArreguez, 2023-03-04:
@jounathaen Looking at the I2C examples on this repository I could not find one simple enough for me to follow - _super new to rust as well_ - so after reading some documentation and looking at example implementations on device libraries I came up with this [example i2c scanner for stm32f1xx](https://github.com/JoaquinEduardoArreguez/stm32f1xx-rust-i2c-scanner), and tried to make code simple and `README` detailed, so anyone can follow it.

I wish I did a better job googling in the first place, since I see now listed the [i2c scanner example in the stm32f3xx-hal](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/examples/i2c_scanner.rs) mentioned here, that should be quite similar and I assume better than what my poorly polished rust skills allowed me to write.