stm32-rs/stm32f1xx-hal #26: LED example toggle behavior

alx741, 2019-02-20:
The LED example turns it on (PC13) briefly and then turns it back off. It happens in this line:

https://github.com/stm32-rs/stm32f1xx-hal/blob/37756dbd0fed43e3175f8c54f0e901b88297cc7f/examples/led.rs#L27

Stepping through it,  it seems to be writing to BSRR (a couple steps after the LED was turned on) turning it off.

Is this a bug or am I missing something?

`Cargo.toml`

```
[dependencies]
cortex-m = "0.5.8"
cortex-m-rt = "0.6.5"
cortex-m-semihosting = "0.3.2"
panic-halt = "0.2.0"
panic-semihosting = "0.5.1"

[dependencies.stm32f1xx-hal]
version = "0.2.0"
features = ["stm32f103", "rt"]
```

TheZoq2, 2019-02-20:
Looks like a bug to me, a recent change changed that line from `into_push_pull_output()` to `into_push_pull_output().set_high()` but the LED is active low.

alx741, 2019-02-20:
ah of course! this is meant for the bluepill board and so the pin must be low in order for the LED to be on

closed 2019-03-08 by TheZoq2