stm32-rs/stm32f1xx-hal #28: Program does not boot without debugger

gbip, 2019-02-21:
Hi !
I can program my stm32f103 using gdb without any troubles, but as soon as I remove the debugger, I can't run my program.

I think that it is correctly flashed, but it does not boot from flash, even if BOOT0 and BOOT1 are set to low.

Do you have any idea to fix this problem ?

TeXitoi, 2019-02-21:
If you have a problem that look like it's working only when the debugger is plugged, there are mainly 2 reasons:
 - you use semihosting, and any printing block your program without a debugger
 - you rely on the debugger cycle counter, and it is not started automatically when the debugger is not plugged.

TeXitoi, 2019-02-21:
Starting the cycle counter: https://github.com/TeXitoi/rusty-clock/blob/master/src/main.rs#L141-L142

gbip, 2019-02-21:
Thank you !!!!!!!!!!!!!!!!!!!

I removed the `cortex_m_semihosting` crate from Cargo.toml and added the two lines you pointed and it works !

therealprof, 2019-02-21:
So I guess we can close this now?

TheZoq2, 2019-02-21:
I don't think you need to remove `cortex_m_semihosting`, you just can't use its printing functionality, right? Or does it only do that?

therealprof, 2019-02-21:
It literally does only do that. There's no other purpose to semihosting...

TeXitoi, 2019-02-21:
there is other usages https://docs.rs/cortex-m-semihosting/0.3.2/cortex_m_semihosting/debug/index.html

But any usage of semihosting will block without a debugger as the system is based of a break point with magic values, and without a debugger, the breakpoint will just stop the program.

gbip, 2019-02-21:
Still, thank you very much TeXitoi for solving this problem, it was a major issue for me !

closed 2019-02-21 by gbip

therealprof, 2019-02-21:
@TeXitoi Oh sure, you can overload what the debugger will do when using the supervisor call...