stm32-rs/stm32f1xx-hal #32: OutputPin trait not satisfied

irwineffect, 2019-02-24:
It appears the embedded_hal::digital::OutputPin trait is not satisfied for stm32f1xx_hal GPIO pins? I'm seeing the following output when compiling:

``` bash
$ cargo build
error[E0277]: the trait bound `stm32f1xx_hal::gpio::gpioa::PA4<stm32f1xx_hal::gpio::Output<stm32f1xx_hal::gpio::PushPull>>: embedded_hal::digital::OutputPin` is not satisfied
  --> src/main.rs:87:24
   |
87 |     let mut max31855 = max31855::Max31855::new(spi, temperature_cs_pin).unwrap();
   |                        ^^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_hal::digital::OutputPin` is not implemented for `stm32f1xx_hal::gpio::gpioa::PA4<stm32f1xx_hal::gpio::Output<stm32f1xx_hal::gpio::PushPull>>`
   |
   = note: required by `<max31855::Max31855<SPI, CS>>::new`

error: aborting due to previous error
```

Full source code:
```rust
#![no_std]
#![no_main]

extern crate panic_halt; // you can put a breakpoint on `rust_begin_unwind` to catch panics

use cortex_m_rt::entry;

use stm32f1xx_hal as hal;
use stm32f1xx_hal::{
    prelude::*,
    pac,
};

use max31855;

#[entry]
fn main() -> ! {
    let c = cortex_m::Peripherals::take().unwrap();
    let p = pac::Peripherals::take().unwrap();

    let mut rcc = p.RCC.constrain();
    let mut flash = p.FLASH.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);
    let mut afio = p.AFIO.constrain(&mut rcc.apb2);
    let mut gpioa = p.GPIOA.split(&mut rcc.apb2);

    // SPI setup
    let spi_sck_pin = gpioa.pa5.into_alternate_push_pull(&mut gpioa.crl);
    let spi_miso_pin = gpioa.pa6;
    let spi_mosi_pin = gpioa.pa7.into_alternate_push_pull(&mut gpioa.crl);
    let spi_mode = hal::spi::Mode {
        polarity: hal::spi::Polarity::IdleLow,
        phase: hal::spi::Phase::CaptureOnFirstTransition
    };

    let mut spi = hal::spi::Spi::spi1(
        p.SPI1,
        (spi_sck_pin, spi_miso_pin, spi_mosi_pin),
        &mut afio.mapr,
        spi_mode,
        100.khz(),
        clocks,
        &mut rcc.apb2
        );

    // temperature sensor driver
    let mut temperature_cs_pin =
        gpioa.pa4.into_push_pull_output_with_state(&mut gpioa.crl,
                                                   stm32f1xx_hal::gpio::State::Low);
    // This doesn't compile!
    let mut max31855 = max31855::Max31855::new(spi, temperature_cs_pin).unwrap();

    loop {}
}
```

burrbull, 2019-02-24:
max31855 depends on old version of embedded_hal.

irwineffect, 2019-02-24:
Thank you, this was the issue! Updating max31855 to be dependent on embedded_hal 0.2.2 fixed it.

closed 2019-02-24 by irwineffect