stm32-rs/stm32f1xx-hal #33: Help Wanted: How to write/read I2C register

jounathaen, 2019-02-24:
Hi,
_Related to #24_ 
As far as my understanding goes, I2C transactions are split into 3 parts (Talking about 7 bit addressing): The device address, the register address and then the data to/from that register (which can be multiple bytes).
Like in this image (from Texas Instruments - SLVA704).
[image]

The functions in this HAL, however, require 2 parameters: address and a byte buffer.
`fn read(&mut self, addr: u8, buffer: &mut [u8]) -> Result<(), Self::Error> `

So here are my two (related) questions:
1.) Which address is meant by the `addr` parameter? The device address or the register address?

2.) How do I write/read a value from -let's say- device 12 register 34? 
Is `addr = 12`, the first element in `buf[0] = 34` and the other bytes in `buf` the data from/to the register in this case?

Thanks and greetings!

burrbull, 2019-02-25:
The function what you need is called
`fn write_read(&mut self, addr: u8, bytes: &[u8], buffer: &mut [u8])`

Where `addr` - 7-bit address of device
first byte of `bytes` - register address to read

How to use:
```
i2c.write_read(self.address, &[reg.addr()], &mut buff)?;
```

jounathaen, 2019-02-25:
Hi, Thanks for the help! I'll try this out today!

Wouldn't it be nice to have a convenience function for this? I guess this is a common I2C use case?
Or maybe rename the function. (`write_read` is IMHO not very self explanatory)

therealprof, 2019-02-25:
@jounathaen I don't see how to make this any more convenient. `write_read` is very self explanatory: first you write, then you read. Your quote from the datasheet is how this particular vendor decided to implement the communication. There're plenty others which work quite a bit differently (from not requiring any register address at all to having a special sequence to figure out how many bytes to read without address).

Anyway, as @burrbull mentioned this interface is defined by the `embedded-hal` crate to allow for writing of universal drivers which seems to be what you want to do. I'd recommend having a look at different drivers to get a better idea of how to do that.

jounathaen, 2019-02-25:
Ok, thanks. Then I had misunderstood the I2C protocol in that case. Thanks for the clarification and the support! I'll try this out and close the issue in the evening!

closed 2019-02-25 by jounathaen