stm32-rs/stm32f1xx-hal #38: Reenable DMA

jamwaffles, 2019-03-07:
DMA was disabled in bae2a2c65c1fdddc6df981a002f455fb232ef611 (by @therealprof), but with no explanation as to why. Can it be reenabled? Does it need some work to get running again? 

I'm happy to try getting it working again, but I have very little experience with DMA.

therealprof, 2019-03-07:
DMA was disabled because the previous experimental implementation does not work on stable (which was the main goal for publication). I think it'll require a bit of rework to make it compile (and work) with stable.

If you can make it work that'd be really cool, though in order to be really useful (e.g. for display drivers) we might have to think about amending the `embedded-hal` traits as well.

TeXitoi, 2019-03-07:
https://docs.rust-embedded.org/embedonomicon/dma.html must also be a great reading to resurect DMA (I didn't read it yet).

jamwaffles, 2019-03-07:
Thanks for the explanation and docs links, I'll attempt some implementation when I have some spare time :slightly_smiling_face:

ssendev, 2019-06-15:
If I read the Changelog right this can be closed since is it was re-enabled

TheZoq2, 2019-06-18:
Yes, that looks correct. Thanks for the reminder :)

closed 2019-06-18 by TheZoq2