stm32-rs/stm32f1xx-hal #42: Timers expose 2 channels and take their pins.

nraynaud, 2019-03-10:
There are a few issues with the current code: 
 - timers only expose 2 channels
 - timers takes the channels pins wether we want it or not.
 - timers force the pin mode to be OpenDrain or PushPull

example form timer_input:
```rust
impl Pins<TIM1> for (PA8<Alternate<OpenDrain>>, PA9<Alternate<OpenDrain>>) {
    const REMAP: u8 = 0b00;
}

```

I propose having the channels pins using the `Option` and a generic mode:
```rust
pub type Pwm1Mapping0<M1, M2, M3, M4> = (Option<PA8<Alternate<M1>>>,
                                         Option<PA9<Alternate<M2>>>,
                                         Option<PA10<Alternate<M3>>>,
                                         Option<PA11<Alternate<M4>>>);
```

here is a more complete example:
https://github.com/japaric/stm32f103xx-hal/blob/9f132cd3a8a95d0b726f925c239fad6e74b60d76/src/pwm2.rs#L219

TheZoq2, 2019-10-07:
This looks like it has been fixed by allowing external impls of the `Pins` trait. This was documented in #60 

However, I feel like it might be more ergonomic to impl all the possible pin/channel combinations, probably via a macro. Is there a reason we haven't done that before?

geomatsi, 2019-10-07:
Hi @TheZoq2 ,
At first I tried to come up with macro, but did not succeeded. So I switched to newtype approach. IIUC some clever macro still would be an option here. My only concern is firmware size: is it possible to list all the combination without bloating the final binary ?

Regards,
Sergey

TeXitoi, 2019-10-08:
Only thé used code Will be on the binary so it shouldn't be a problem.

TheZoq2, 2019-10-10:
I'll try to experiment with this when I get time

TheZoq2, 2019-12-02:
This has been fixed in #147

closed 2019-12-02 by TheZoq2