stm32-rs/stm32f1xx-hal #66: SPI can become unresponsive in some conditions

BenBergman, 2019-05-18:
I was attempting to write an SPI driver for ST7789 based displays using a Blue Pill as my base board. My screen was unresponsive so I used a logic analyzer to verify what was being sent out by my board. I noticed that the board was still toggling the LCD's DC pin just fine (and blinking an LED), but the SPI clock and data lines became completely silent after the first two bytes. The first two bytes were correct though.

What is weird is that I have also connected an SSD1306 SPI OLED display and it runs just fine. They use different SPI modes though, so that might be related.

I have attached my code as well as the logic probe data (collected through Saleae software) for both the non-functioning Blue Pill code as well as very similar code running from an Arduino library which works exactly as expected.

[debug_rust_st7789_driver.zip](https://github.com/stm32-rs/stm32f1xx-hal/files/3193607/debug_rust_st7789_driver.zip)

I have since also copied my rust code over to an `atsamd` based project and it runs just fine there, so this leads me to believe there is a bug in this crate.

Let me know if there is any more information I can provide. I have borrowed the probe temporarily, so if any more data collection is needed let me know soon so I can ensure I still have access.

mvirkkunen, 2019-05-25:
If I'm reading this right, I think you're using the non-blocking `send` without waiting that the transfer is complete. Try using the blocking `write` from `embedded_hal::blocking::spi` instead. It can also write entire buffers without having to write each byte as a separate call. Also, all the methods can return errors, so it would be a good idea to handle those.

BenBergman, 2019-05-27:
Aha, switching to `write` fixes it for the blue pill. Curious that `send` worked for one chip and not another, but I guess that boils down to the way the `hal` is implemented for each.

Thanks for the help!

mvirkkunen, 2019-05-27:
That's possible, but in the original code each `send` caused a race condition, so it's possible that the timing just worked out correctly by chance on another chip. At any rate `send` is allowed to fail if it would block.

therealprof, 2019-05-28:
@BenBergman I guess this can be closed then?

BenBergman, 2019-05-28:
Yeah, sounds like I was just using it wrong. Thanks for the help!

closed 2019-05-28 by BenBergman