stm32-rs/stm32f1xx-hal #70: Deprecation warnings for use of embedded_hal::digital::v1 traits

sapir, 2019-06-03:
I'm getting deprecation warnings for using the GPIO pins, like:
```
warning: use of deprecated item 'embedded_hal::digital::v1::OutputPin::set_high': Deprecated because the methods cannot return errors. Users should use the traits in digital::v2.
```

TheZoq2, 2019-06-06:
This is certainly a problem, @therealprof you "Not sure we really want to deal with the fall out and actually how to deal with it." in #63, but it feels like we have to do it at some point. Or are there more changes to the digital api planned?

therealprof, 2019-06-06:
No idea, I've not been a big fan of this whole `digital::v2` mess to begin with but the jury is still out how to properly address the fallout. 

CC https://github.com/rust-embedded/embedded-hal/issues/139
CC https://github.com/rust-embedded/embedded-hal/issues/135

TeXitoi, 2019-11-16:
That's fixes, no?

Feel free to reopen if that's not the case.

closed 2019-11-16 by TeXitoi