stm32-rs/stm32f1xx-hal #76: blink example does not compile

MauroMombelli, 2019-07-07:
Hello,
new in rust.
Following the readme of the project, i managed to set up the toml to make it work:

```
[package]
name = "xxxx"
version = "0.1.0"
authors = ["xxxx"]
edition = "2018"

[profile.release]
# optimize for size ('z' would optimize even more)
opt-level = 's'
# link with link time optimization (lto).
lto = true
# enable debugging in release mode.
debug = true

[dependencies]
cortex-m = "0.6.0"
nb = "0.1.2"
cortex-m-rt = "0.6.8"
stm32f1 = "0.7.1"

[dependencies.stm32f1xx-hal]
version = "0.2.0"
features = ["stm32f100", "rt"]

```

using 0.2.0 as requested from the readme return the error:

```
error[E0609]: no field `pc13` on type `stm32f1xx_hal::gpio::gpioc::Parts`
```

then i notice 0.3.0 is available, and with that one i get warning about digital::v1 (I read a little bit the issue by my skill level are 0 and i could not understand nothing).

Please please please fix your own doc and warning,
sincerely,

a frustrated beginner

MauroMombelli, 2019-07-07:
a kind stranger told me the fix should simply be changint to hal::digital::V2::{InputPin, OutputPin, StatefulOutputPin, toggleable}; line 61 of https://docs.rs/stm32f1xx-hal/0.2.1/src/stm32f1xx_hal/gpio.rs.html#61

TeXitoi, 2019-07-08:
Don't bother too much about this deprecation warning for the moment. The following issue is #70 

The README should be updated.

MauroMombelli, 2019-07-08:
>The README should be updated.

so it is fine that 0.2.0 doe NOT compile? unfortunately the example seems to be spread also in the crates.io and I think I saw it also in another official guide, could it be the cortex-m guide? If i find it again I will update this issue so at least there will be a reference

closed 2019-07-08 by therealprof

TeXitoi, 2019-07-08:
0.2 is not API compatible with 0.3. I suspect the code you've tried was for 0.3, and thus didn't compile with 0.2, or some other crate version missmatch problem.

If you want that I check more in depth why it doesn't compile for 0.2, I will need a complete example.

0.2 is not supposed to be used anymore anyway, so that's more by curiosity or to learn.

MauroMombelli, 2019-07-08:
the blink code that was in the readme and in the crate.io (it still report to use 0.2.0 with the non-compile example, here https://crates.io/crates/stm32f1xx-hal)

TeXitoi, 2019-07-08:
In 0.2.1, PC13 was not declared for stm32f100: https://github.com/stm32-rs/stm32f1xx-hal/commit/49ee9cb4175a531a39fc4c58c62d077e690c1d00#diff-73928a829df74085c37fe73a4f3484fc

PR #78 (2019-07-08): Update the cargo version in the README.md
fixes #76
commit 32d50f8c (2019-07-08): Update the cargo version in the README.md

fixes #76