stm32-rs/stm32f1xx-hal #80: timer interrupt not called

MauroMombelli, 2019-07-11:
This is a request for help, not a bug report.
Hope this is fine, from the readme looks like this is the place to ask. (also not sure if an issue of this crate or cortex_m or cortex_m_rt)

My issue: after managing to run the blink, the step N°2 is the "BlinkWithoutDelay".

the pseudocode:
```
last = 0
loop:
  if millis() - last > 1000:
    last = millis()
    led.toggle()
```

so my final (non working) code (you can see i am trying to use TIM2, but also tried with SYST with no success too)

```
//#![deny(warnings)] //  If the Rust compiler generates a warning, stop the compilation with an error.
#![deny(unsafe_code)]  //  Don't allow unsafe code in this file.
#![no_std]  // std and main are not available for bare metal software
#![no_main] 

use cortex_m_rt::{entry};
//use cortex_m_rt::{entry, exception, ExceptionFrame};
use stm32f1xx_hal::device::interrupt;
use nb::block;

use stm32f1xx_hal::{
    prelude::*,
    pac,
    timer::{Timer,Event},
    serial::{Serial},
};

use core::sync::atomic::{AtomicUsize, Ordering};

static COUNTER_MS: AtomicUsize = AtomicUsize::new(0);

#[entry]
fn main() -> ! {
    // Get access to the core peripherals from the cortex-m crate
    let mut cp = cortex_m::Peripherals::take().unwrap();
    // Get access to the device specific peripherals from the peripheral access crate
    let dp = pac::Peripherals::take().unwrap();

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs
    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();
    let mut afio = dp.AFIO.constrain(&mut rcc.apb2);

    //free pin PB3, PB4 and PA15
    afio.mapr.disable_jtag();

    // Freeze the configuration of all the clocks in the system and store
    // the frozen frequencies in `clocks`
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Acquire the GPIO peripheral
    let mut gpioa = dp.GPIOA.split(&mut rcc.apb2);
    let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);

    let mut led1 = gpiob.pb3.into_push_pull_output(&mut gpiob.crl);
    let mut led2 = gpiob.pb4.into_push_pull_output(&mut gpiob.crl);
    led2.set_high();
    
    led1.set_high();

    //let mut timer = Timer::syst(cp.SYST, 1_000.hz(), clocks);
    //timer.listen(Event::Update);
    
    let mut timer = Timer::tim2(dp.TIM2, 1_000.hz(), clocks, &mut rcc.apb1);
    timer.listen(Event::Update);
    
    // USART1
    let txp = gpioa.pa9.into_alternate_push_pull(&mut gpioa.crh);
    let rxp = gpioa.pa10;
    let serial = Serial::usart1(
        dp.USART1,
        (txp, rxp),
        &mut afio.mapr,
        9_600.bps(),
        clocks,
        &mut rcc.apb2,
    );

    led1.set_low();

    // Split the serial struct into a receiving and a transmitting part
    let (mut tx, mut rx) = serial.split();

    // Wait for the timer to trigger an update and change the state of the LED
    let mut timer_start = 0;
    
    loop {
        
        let now = COUNTER_MS.load(Ordering::Relaxed);
        
        let diff = now - timer_start;
        if diff > 1_000 {
            timer_start = now;
            led1.toggle();
        }
        block!(tx.write(now as u8)).ok();
        
        if let Ok(ret) = rx.read() {
            led1.toggle();
            cortex_m::asm::delay(5_000_000);
            led1.toggle();
            cortex_m::asm::delay(5_000_000);
            led1.toggle();
            cortex_m::peripheral::SCB::system_reset(&mut cp.SCB);
        }
    }

}

#[interrupt]
fn TIM2() {
    COUNTER_MS.fetch_add(1, Ordering::Relaxed);
}

/*
#[exception]
fn SysTick() {
    COUNTER_MS.fetch_add(1, Ordering::Relaxed);
}
*/

#[panic_handler]
fn my_panic(_info: &core::panic::PanicInfo) -> ! {
    loop {
    
        let dp = pac::Peripherals::take().unwrap();
        let mut rcc = dp.RCC.constrain();
        let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);
        let mut led2 = gpiob.pb4.into_push_pull_output(&mut gpiob.crl);
        led2.toggle(); //is inverted logic
        cortex_m::asm::delay(100_000_000);
    }
}

```

the usart and reset is there for debug purpose

geomatsi, 2019-07-12:
In brief:
* you missed quite a few bits related to NVIC setup
* TIM2 needs to be restarted  in its interrupt handler
* SYST API is different

Here is an example that uses several timers including SysTick: [code](https://github.com/geomatsi/e.ziclean-cube/blob/master/examples/hal/example-hal-timer-irq.rs). It is for stm32f101, but should be working fine on BluePill as well.

MauroMombelli, 2019-07-13:
thanks, I though the interrupt setup was handles by the timer class (and this is also why i thought syst and timX would have the same API)

Why would the timer require to be restarted? only one shot mode is enabled?

MauroMombelli, 2019-07-13:
thanks, I though the interrupt setup was handles by the timer class (and this is also why i thought syst and timX would have the same API)

Why would the timer require to be restarted? only one shot mode is enabled?

I tried to use the sysclok has you suggested, but my code just reset and I am not sure what I am doing wrong

```
//#![deny(warnings)] //  If the Rust compiler generates a warning, stop the compilation with an error.
#![deny(unsafe_code)]  //  Don't allow unsafe code in this file.
#![no_std]  // std and main are not available for bare metal software
#![no_main] 

use nb::block;

use stm32f1xx_hal::{
    prelude::*,
    pac,
    timer::{Timer,Event},
    serial::{Serial},
};

use core::sync::atomic::{AtomicUsize, Ordering};

static COUNTER_MS: AtomicUsize = AtomicUsize::new(0);

#[cortex_m_rt::entry]
fn main() -> ! {
    // Get access to the core peripherals from the cortex-m crate
    let mut cp = cortex_m::Peripherals::take().unwrap();
    // Get access to the device specific peripherals from the peripheral access crate
    let dp = pac::Peripherals::take().unwrap();

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs
    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();
    let mut afio = dp.AFIO.constrain(&mut rcc.apb2);

    //free pin PB3, PB4 and PA15
    afio.mapr.disable_jtag();

    // Freeze the configuration of all the clocks in the system and store
    // the frozen frequencies in `clocks`
    let clocks = rcc.cfgr.sysclk(8.mhz()).pclk1(8.mhz()).freeze(&mut flash.acr);

    // Acquire the GPIO peripheral
    let mut gpioa = dp.GPIOA.split(&mut rcc.apb2);
    let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);

    let mut led1 = gpiob.pb3.into_push_pull_output(&mut gpiob.crl);
    let mut led2 = gpiob.pb4.into_push_pull_output(&mut gpiob.crl);
    led2.set_high();
    
    led1.set_high();

    // configure SysTick timer
    let mut syst = cp.SYST;
    syst.set_clock_source(cortex_m::peripheral::syst::SystClkSource::Core);
    syst.set_reload(16_000_000 - 1);
    syst.enable_counter();
    syst.enable_interrupt();
    
    // USART1
    let txp = gpioa.pa9.into_alternate_push_pull(&mut gpioa.crh);
    let rxp = gpioa.pa10;
    let serial = Serial::usart1(
        dp.USART1,
        (txp, rxp),
        &mut afio.mapr,
        9_600.bps(),
        clocks,
        &mut rcc.apb2,
    );

    led1.set_low();

    // Split the serial struct into a receiving and a transmitting part
    let (mut tx, mut rx) = serial.split();

    // Wait for the timer to trigger an update and change the state of the LED
    let mut timer_start = 0;
    
    loop {
        
        let now = COUNTER_MS.load(Ordering::Relaxed);
        
        let diff = now - timer_start;
        if diff > 1_000 {
            timer_start = now;
            led1.toggle();
        }
        
        block!(tx.write(now as u8)).ok();

    }

}

#[cortex_m_rt::exception]
fn SysTick() {
    COUNTER_MS.fetch_add(1, Ordering::Relaxed);
}

#[panic_handler]
fn my_panic(_info: &core::panic::PanicInfo) -> ! {
    loop {
    
        let dp = pac::Peripherals::take().unwrap();
        let mut rcc = dp.RCC.constrain();
        let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);
        let mut led2 = gpiob.pb4.into_push_pull_output(&mut gpiob.crl);
        led2.toggle(); //is inverted logic
        cortex_m::asm::delay(100_000_000);
    }
}
```

MauroMombelli, 2019-07-13:
same issue trying with TIM2..
I am really lost in what could be the issue, full code:

```
//#![deny(warnings)] //  If the Rust compiler generates a warning, stop the compilation with an error.
#![deny(unsafe_code)]  //  Don't allow unsafe code in this file.
#![no_std]  // std and main are not available for bare metal software
#![no_main] 

use cortex_m_rt::{entry};
//use cortex_m_rt::{entry, exception, ExceptionFrame};
use stm32f1xx_hal::device::interrupt;
use nb::block;

use stm32f1xx_hal::{
    prelude::*,
    pac,
    timer::{Timer,Event},
    serial::{Serial},
};

use core::sync::atomic::{AtomicUsize, Ordering};

static COUNTER_MS: AtomicUsize = AtomicUsize::new(0);

#[entry]
fn main() -> ! {
    // Get access to the core peripherals from the cortex-m crate
    let mut cp = cortex_m::Peripherals::take().unwrap();
    // Get access to the device specific peripherals from the peripheral access crate
    let dp = pac::Peripherals::take().unwrap();

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs
    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();
    let mut afio = dp.AFIO.constrain(&mut rcc.apb2);

    //free pin PB3, PB4 and PA15
    afio.mapr.disable_jtag();

    // Freeze the configuration of all the clocks in the system and store
    // the frozen frequencies in `clocks`
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Acquire the GPIO peripheral
    let mut gpioa = dp.GPIOA.split(&mut rcc.apb2);
    let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);

    let mut led1 = gpiob.pb3.into_push_pull_output(&mut gpiob.crl);
    let mut led2 = gpiob.pb4.into_push_pull_output(&mut gpiob.crl);
    led2.set_high();
    
    led1.set_high();

    let mut timer = Timer::tim2(dp.TIM2, 1_000.hz(), clocks, &mut rcc.apb1);
    timer.listen(Event::Update);
    
    let nvic = &mut cp.NVIC;
    nvic.enable(stm32f1xx_hal::stm32::Interrupt::TIM2);
    cortex_m::peripheral::NVIC::unpend(stm32f1xx_hal::stm32::Interrupt::TIM2);
    
    // USART1
    let txp = gpioa.pa9.into_alternate_push_pull(&mut gpioa.crh);
    let rxp = gpioa.pa10;
    let serial = Serial::usart1(
        dp.USART1,
        (txp, rxp),
        &mut afio.mapr,
        9_600.bps(),
        clocks,
        &mut rcc.apb2,
    );

    led1.set_low();

    // Split the serial struct into a receiving and a transmitting part
    let (mut tx, mut rx) = serial.split();

    // Wait for the timer to trigger an update and change the state of the LED
    let mut timer_start = 0;
    
    loop {
        
        let now = COUNTER_MS.load(Ordering::Relaxed);
        
        
        let diff = now - timer_start;
        if diff > 1_000 {
            timer_start = now;
            led1.toggle();
        }
        
        block!(tx.write(now as u8)).ok();
    }

}

#[interrupt]
fn TIM2() {
    COUNTER_MS.fetch_add(1, Ordering::Relaxed);
    let dp = pac::Peripherals::take().unwrap();
    let mut rcc = dp.RCC.constrain();
    let mut flash = dp.FLASH.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);
    let mut timer = Timer::tim2(dp.TIM2, 1_000.hz(), clocks, &mut rcc.apb1);
    timer.start(1.hz());
}

#[panic_handler]
fn my_panic(_info: &core::panic::PanicInfo) -> ! {
    loop {
    
        let dp = pac::Peripherals::take().unwrap();
        let mut rcc = dp.RCC.constrain();
        let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);
        let mut led2 = gpiob.pb4.into_push_pull_output(&mut gpiob.crl);
        led2.toggle(); //is inverted logic
        cortex_m::asm::delay(100_000_000);
    }
}
```

therealprof, 2019-07-13:
What's up with all the setup in the interrupt and panic handler? You can call `pac::Peripherals::take().unwrap()` exactly **once**.

And no, you the interrupts are not set up automatically. You need to do something along the lines of https://github.com/stm32-rs/stm32f0xx-hal/blob/master/examples/led_hal_button_irq.rs#L74-L78 to enable the interrupt handlers.

MauroMombelli, 2019-07-13:
> You can call pac::Peripherals::take().unwrap() exactly once.

why it let me do it? I read a couple of guides but never saw this pointer out, so I just did it

> And no, you the interrupts are not set up automatically

yep, see it and fixed in the next examples.

So i guess not my crash are cause by those double pac::Peripherals in my TIM2 example, but teh syst example should run since the double take would happen only in case of hardfault, witch means something already went very wrong.

I am sorry if I am pedantic, but I really think those stuff are what other beginner will have to deal with, and I want to leave a public trace of how to deal with that, maybe in the end collect it in some small examples and added to the documentation

TeXitoi, 2019-07-14:
Peripheral is a global `Option`, thus, the first `.take().unwrap()` will work, the second will panic. Being able to have the peripherals only once is what allow the rust type system to manage the state of the pins.

thalesfragoso, 2019-07-18:
> thanks, I though the interrupt setup was handles by the timer class (and this is also why i thought syst and timX would have the same API)

Syst is from the cortex-m core itself, the other timers are vendor specific, they are bound to be different. We have an almost identical API in this HAL for both, there are just somethings that you must remember:

- If you are using vendor timers you need to configure the NVIC part yourself, i.e. enable interrupts, set priorities and maybe first unpend them once for good measure. We don't do this in the HAL itself because we don't have exclusive access to the NVIC registers, so it can't be done safely. You also need to clear interrupt flag before exiting the interrupt handler, otherwise it will just fire again indefinitely, the HAL provides methods for that.
- The reload value method for Syst takes an u32, but it is actually an u24, this is documented in the manuals and also in `cortex-m` crate.

Maybe they are other nuances that I do not recall right now, please correct me if that it is the case.

> Why would the timer require to be restarted? only one shot mode is enabled?

It does not.

> why it let me do it? I read a couple of guides but never saw this pointer out, so I just did it

This is clearly stated in the documentation, see : https://docs.rs/cortex-m/0.6.0/cortex_m/peripheral/index.html

There is a way around this, but it is highly unsafe and you could break 99% of this HAL (and other things) if you do that wrong, so it is not recommend unless you are entirely sure of what you are doing.

>So i guess not my crash are cause by those double pac::Peripherals in my TIM2 example, but teh syst example should run since the double take would happen only in case of hardfault, witch means something already went very wrong.

First, there is a minor problem in your initialization, from `cortex-m` docs:

> NOTE The reference manual indicates that:
>
>"The SysTick counter reload and current value are undefined at reset, the correct initialization sequence for the SysTick counter is:
>
>    Program reload value
>    Clear current value
>    Program Control and Status register"
>
>The sequence translates to self.set_reload(x); self.clear_current(); self.enable_counter()

Second, and probably most important, you are setting the reload value to `16_000_000 - 1`, with a 8Mhz clock this translates to a two seconds period, and then in the loop you check to see if you variable was increment one thousand times. So maybe your code is working okay, you just did not wait 2000 seconds to see the first blink.

TheZoq2, 2019-10-07:
Since this was a call for help rather than a bug report and it seems quiet now, i'll go ahead and close it. Feel free to re-open if things change!

closed 2019-10-07 by TheZoq2