stm32-rs/stm32f1xx-hal #82: adc example issue

yhyuan, 2019-07-14:
I am trying to run this adc example. but I found this line of code failed to compile.
```
let mut adc = adc::Adc::adc1(p.ADC1, &mut rcc.apb2, clocks);
```
The error message is 
```
error[E0061]: this function takes 2 parameters but 3 parameters were supplied
  --> src/main.rs:28:19
   |
28 |     let mut adc = adc::Adc::adc1(p.ADC1, &mut rcc.apb2, clocks);
   |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected 2 parameters

error: aborting due to previous error

For more information about this error, try `rustc --explain E0061`.
error: Could not compile `blue-pill-quickstart`.
```
It seems the api has been changed.

TheZoq2, 2019-07-14:
Yes, that change was made in #67 which has not yet been published to crates.io which is why the example fails to build.

Until we publish it, you can have a look at the examples in the 0.3.0 tag https://github.com/stm32-rs/stm32f1xx-hal/tree/v0.3.0

yhyuan, 2019-07-15:
> Yes, that change was made in #67 which has not yet been published to crates.io which is why the example fails to build.
> 
> Until we publish it, you can have a look at the examples in the 0.3.0 tag https://github.com/stm32-rs/stm32f1xx-hal/tree/v0.3.0

Thanks, I guess I can use the github to directly use the current master branch rather than use the one in crates.io.

TheZoq2, 2019-07-16:
That works as well. I'll go ahead and close this issue since it will be fixed when we publish the next version

closed 2019-07-16 by TheZoq2