stm32-rs/stm32f1xx-hal #84: HPRE calculated incorrectly

kyrias, 2019-07-15:
Reading through [RM0008](https://www.st.com/resource/en/reference_manual/cd00171190.pdf), it seems like HPRE is being calculated incorrectly [here](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/rcc.rs#L191-L204).  The calculation works up until division by 16, after that the reference manual skips to 64 whereas this crate continues with 32, so you get the following correspondence:

hpre_bits | Reference manual | Crate
-|-|-
0xxx | SYSCLK not divided | `(1 << (0b0111 - 0b0111)) == 1`
1000 | SYSCLK divided by 2 | `(1 << (hpre_bits - 0b0111)) == 2`
1001 | SYSCLK divided by 4 | `(1 << (hpre_bits - 0b0111)) == 4`
1010 | SYSCLK divided by 8 | `(1 << (hpre_bits - 0b0111)) == 8`
1011 | SYSCLK divided by 16 | `(1 << (hpre_bits - 0b0111)) == 16`
1100 | SYSCLK divided by 64 | `(1 << (hpre_bits - 0b0111)) == 32`
1101 | SYSCLK divided by 128 | `(1 << (hpre_bits - 0b0111)) == 64`
1110 | SYSCLK divided by 256 | `(1 << (hpre_bits - 0b0111)) == 128`
1111 | SYSCLK divided by 512 | `(1 << (hpre_bits - 0b0111)) == 256`

kyrias, 2019-07-15:
What's sort of devious about this is that `ppre1` and `ppre2` are set to the wrong frequencies, but `pclk1` and `pclk2` end up being calculated to the requested frequency values due to using the same kind of incorrect logic as `hclk`, so the software thinks that the clocks were set properly.

It's a bit tricky to work through the logic, so here's an example where it goes wrong:

```
sysclk                       = 72_000_000
wanted_hclk   = sysclk / 128 =    562_500
wanted_pclk1  = hclk / 4     =    140_625
wanted_pclk2  = hclk / 2     =    281_250

sysclk / wanted_hclk = 128  =>  hpre_bits := 0b1101  # Correct, 0b1101 is divide by 128, which sets HCLK to 562_500.
hclk :=     sysclk / (1 << (hpre_bits - 0b0111))
     := 72_000_000 / (1 << (0b1101    - 0b0111))  => 1_125_000  # We set hardware to 562_500 but
                                                                # tell software that it's 1_125_000.

hclk / wanted_pclk1 = 8  => ppre1_bits := 0b110  # We wanted divide by 4, but got divide by 8. This means
                                                 # that PCLK1 will be set to 70_312.5, instead of 140_625.
pclk1 :=      hclk / (1 << (ppre1_bits - 0b011))
      := 1_125_000 / (1 << (0b110      - 0b011)) => 140_625  # Even though we set the clock to 70_312.5, we're telling
                                                             # software that it was set to 140_625 as requested.
```

thalesfragoso, 2019-07-16:
Nice catch, take a look at https://github.com/stm32-rs/stm32f1xx-hal/pull/85, I think it solves it. Thanks.

kyrias, 2019-07-16:
Another thing that I'm unsure about is cases where the `hclk` divided by a given `pclk1` doesn't give you a round quotient, and ends up rounding down.

As an example, if you specify:
* `hclk` of `64_000_000` (maximum `sysclk` when `HSI` is the `PLL` input)
* `pclk1` of `36_000_000` (which is the maximum `pclk1` speed)

You get `hclk / pclk1 = 1.77...`, which gets rounded down to `1`, and then we panic because `64_000_000` divided by `1` is greater than `36_000_000`.  So it really feels to me that this division should be done as floats, and then round up before converting to a u32, so we never try to exceed the known maximum limits.

If the clock settings to be written to the hardware could be calculated in a `const fn`, or as part of a macro, it would even be possible to have it be a compile-time error if you try to specify impossible-to-achieve clock speeds.  With https://github.com/rust-lang/rust/issues/54140 it would be possible to make it a compile-time warning instead if going the macro route, hm.

closed 2019-07-17 by therealprof