stm32-rs/stm32f1xx-hal #87: ERROR(cortex-m-rt): The interrupt vectors are missing.

PvdBerg1998, 2019-07-18:
I'm getting this error when linking with Rusts default LLD linker. Switching to `"-C", "linker=arm-none-eabi-ld"` fixes the problem.

TheZoq2, 2019-07-18:
Did you forget to post the error you get?

PvdBerg1998, 2019-07-18:
It's in the title. I should probably have posted some more information though. Here you go:
```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" <snip> "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
```

Oddly enough the crate uses the rust LLD linker if I'm not mistaken.

therealprof, 2019-07-18:
What exactly are you trying to compile? Is there a crate somewhere? Is it an example? Which Rust version are you using? Without a way to reproduce it there's very little chance of fixing it...

PvdBerg1998, 2019-07-18:
It's not an example, though it should be exactly the same as the quickstart. I started rewriting a recent project of mine from STM32F103. Running latest nightly on win10.

therealprof, 2019-07-18:
Oh, nightly is broken for thumb targets at the moment. Not sure what the problem is but it seems to be related to the LLVM9 upgrade...

PvdBerg1998, 2019-07-18:
Ah, sorry for the false positive then. Thanks for the great work!

closed 2019-07-18 by PvdBerg1998