stm32-rs/stm32f1xx-hal #89: Add `count` method to `Timer` struct

mjepronk, 2019-07-27:
In order to create a reusable timer which we can use to handle timeouts[1] in our code, I would like to have an additional method `count` to the `Timer` struct. This would return the value of the `cnt` register. This method already exists for the QEI. User code would look something like this:
```rust
// Timer of 1,000 Hz, so once per ms
let timer = Timer::tim2(tim2, Hertz(1_000), clocks, apb1);
timer.start(Hertz(1_000)); // the counter should be reset here
// Some other code...
if timer.count() > 500 {
  // More than 500 ms have passed...
}
```
It's easier than using the timer's interrupt. Or am I missing something? It may also be useful to expose a `reset()` method that resets the counter. I've noticed that resetting a timer requires unsafe code...

[1] My use case is a timer that I can use for timeouts when implementing a serial protocol (in this case to talk to the ESP-01 in AT mode).

TheZoq2, 2019-07-28:
I don't see a reason for not having this, feel free to submit a PR, and if you're not sure how to implement it, let me know!

> It's easier than using the timer's interrupt. Or am I missing something?

Absolutely, even though it's slightly less performant

> It may also be useful to expose a reset() method that resets the counter.

The timer is already reset by calling `start` if I recall correctly. I suppose having a dedicated reset method might also be useful, though it would require the timer storing the time that was set previously.

> I've noticed that resetting a timer requires unsafe code...

It does, and I believe it's because there is no way for the svd crate to know what values are allowed in the register. In the case of a timer, any value can be set without issues.

burrbull, 2019-08-07:
Is there any reason to start timer in init. I think start must be explicit.
```
    let timer = Timer::tim2(tim2, clocks, apb1);
    timer.start(1_000.hz());
```

TheZoq2, 2019-08-10:
I too find that pretty weird. The one reason I can think of is that you avoid undefined behaviour when calling `timer.wait()` before calling `timer.start`

burrbull, 2019-08-10:
Maybe use state machine with 2 states?
```rust
struct Timer; // start
struct TimerStarted; // wait, stop, etc.
```

burrbull, 2019-08-10:
But this will be incompatible with `embedded-hal`. It's bad.

burrbull, 2019-08-10:
By the way, what happens if we call `wait` after `stop`?

TheZoq2, 2019-08-11:
> But this will be incompatible with embedded-hal. It's bad.

Good point

> By the way, what happens if we call wait after stop?

I think it would freeze, which seems like a bad thing.

closed 2019-08-31 by TheZoq2