stm32-rs/stm32f1xx-hal #102: GPIO output toggling does not seem to be working

dbrgn, 2019-08-18:
I'm using the blue pill board, which has an LED connected to PC13.

With the following code, the LED lights up as soon as the pin is configured as push/pull output, but it cannot be turned off (set_low() does not turn off the LED, and for sanity checking set_high does not either).

```rust
#![deny(unsafe_code)]
#![no_main]
#![cfg_attr(not(test), no_std)]

// Panicking behavior
extern crate panic_semihosting; // logs messages to the host stderr; requires a debugger

#[macro_use]
extern crate cortex_m_rt;

use cortex_m_semihosting::hprintln;
use embedded_hal::digital::v2::OutputPin;
use stm32f1xx_hal::{prelude::*, pac};

#[entry]
fn main() -> ! {
    hprintln!("init").unwrap();

    let device: pac::Peripherals = pac::Peripherals::take().unwrap();

    let mut rcc = device.RCC.constrain();
    let mut gpioc = device.GPIOC.split(&mut rcc.apb2);

    let mut led = gpioc.pc13.into_alternate_push_pull(&mut gpioc.crh);
    led.set_low().unwrap();

    hprintln!("loop").unwrap();
    loop {
    }
}
```

This code uses stm32f1 0.8 and stm32f1xx-hal 0.4. I also tried stm32f1 0.6 and stm32f1xx-hal 0.2, which showed the same behavior.

Any idea what the problem could be?

The full code can be found [here](https://github.com/dbrgn/desklift/tree/led-bug/firmware) (branch `led-bug`).

TheZoq2, 2019-08-18:
Hmm, I believe the issue is that you're calling `into_alternate_push_pull` when you should be calling `into_push_pull_output`. Alternate push_pull is for pins which are used by other devices, such as the UART or SPI peripheral, not for general use.

With that said, I have no idea why we implement `set_low()` and `set_high()` on those, @therealprof do you happen to know?

dbrgn, 2019-08-18:
Thanks, that's indeed the problem. Combined with the fact that I didn't realize that the pin sinks the current by default instead of sourcing it:

[image]

(But I *did* try `set_high`, which didn't work.)

By using `into_push_pull_output` + `set_high` the LED turns off. Same thing happens when using `into_push_pull_output_with_state` with initial state `High`.

I guess this issue can be closed, unless @therealprof agrees that `set_low` and `set_high` should not be implemented for the alternate function pins :slightly_smiling_face:

TheZoq2, 2019-08-18:
Great to hear! I'll leave it open until we decide if this should be fixed :)

therealprof, 2019-08-18:
I vaguely remember some issue which required that for something but I can't find anything useful at the moment. I'd say let's remove it.

PR #103 (2019-08-18): Remove OutputPin impl for Alternate mode pins
Closes #102
closed 2019-08-18 by therealprof