stm32-rs/stm32f1xx-hal #110: impl<USART> core::fmt::Write for Tx<USART> seems to be removed without ?

ijager, 2019-09-26:
In commit d66f170bac185dbbf07a88c2ff0a42f4d0b46ac5 support for string formatting was added to serial Tx. But then a month later in a seemingly unrelated commit 66cfb785279dc9c333a4800abe231d07158f1aa4 about DMA it was removed again, [here](https://github.com/stm32-rs/stm32f1xx-hal/commit/66cfb785279dc9c333a4800abe231d07158f1aa4#diff-28d049255be5c019ebb0714798c77ff0L548).

It seems like an accident, since there is no mention about it in the commit message. Or was there a valid reason?

Adding it back makes it work again at least.

TheZoq2, 2019-09-27:
That does look odd, and I don't see a reason for having removed it either. @therealprof or @burrbull perhaps you have more insight since you authored and merged the relevant PR.

@ijager feel free to submit a PR that re-adds it! Alternatively, I can try get it done this weekend if I remember

therealprof, 2019-09-27:
I blame accidents and sloppy reviewing. Sorry. 😐

burrbull, 2019-09-27:
I can't remember I remove something from serial. Maybe did not see this when rebase or merge.

closed 2019-09-29 by TheZoq2