stm32-rs/stm32f1xx-hal #122: spurious character in file src/i2c.rs

pdgilbert, 2019-10-16:
There is an error using
cargo generate --name stm32f1xx-hal --git https://github.com/stm32-rs/stm32f1xx-hal

     let sr1 = $i2c.sr1.read();
Error: liquid:    --> 194:9
    |
194 |         let sr1 = $i2c.sr1.read();\u240a
    |         ^---
    |
    = expected Literal

This line seems to be in the file src/i2c.rs.  It has an (almost) invisible \u240a character at the end.

therealprof, 2019-10-16:
Hm, weird. I just checked and there's nothing in the current git master.

pdgilbert, 2019-10-16:
I assumed the character was the cause of the error  reported by 'cargo generate', but it may be that the character is format wrapper added by cargo generate when reporting another error. (It seems the character is neither in a git checkout nor in the file generated by cargo generate.)
Also, I cannot generate the error with cargo build or cargo check.

So I should not have assumed the cause,  the issue really is that  'cargo generate' gives
 Error: liquid: --> 194:9
|
194 | let sr1 = $i2c.sr1.read();\u240a  
 | ^---
|
= expected Literal

burrbull, 2019-10-18:
`cargo generate` require specially formed (binary) repository. `stm32f1xx-hal` is library.

https://github.com/TeXitoi/blue-pill-quickstart/issues/7

TheZoq2, 2019-10-23:
I guess that explains that. Kind of weird behaviour in my opinion, but I guess that's the fault of `cargo generate`

closed 2019-10-23 by TheZoq2