stm32-rs/stm32f1xx-hal #123: Possible to use peripheral in panic handler?

riaqn, 2019-10-19:
Hi, first thank you for the effort put into this wonderful ecosystem.

I'm writing a heater controller with this library, and one thing I want to make sure of, is that the PWM (which controls how much heat is produced) is disabled/zeroed when panic happens, so that it doesn't burned down my house.

From what I gathered, peripherals are intentionally made hard to get for safety reasons. I can't `take` in the handler because that won't work. Maybe I should try global mutable variable to store the PWM instance, but it's considered a bad practice in rust?

TheZoq2, 2019-10-19:
Since I pressume you're not going to do anything after the panic handler, you can use the [steal] function on the periperal, re-setup the peripherals and turn off the PWM. Alternatively, you can probably just call reset on the whole thing after stealing.

[steal]: https://docs.rs/stm32f1/0.8.0/stm32f1/stm32f103/struct.Peripherals.html#method.steal

riaqn, 2019-10-19:
@TheZoq2 thanks for the reply.

Re-setup sounds like a hassle, I'm especially worried about re-initializing these registers causing some undefined behaviour. 

Is there someway I can directly access the PWM object that I constructed in `main`?

TheZoq2, 2019-10-19:
A global PWM variable is probably the best option. It's not "best practice", but it's also not terrible because you'll have to access it via unsafe

riaqn, 2019-10-19:
Thanks!  I tried this with the following program. But obviously my borrow-fu is not very good yet.

```
static mut pwm_ : Option<&mut Pwm<TIM3, C1>> = None;
#[panic_handler]
unsafe fn panic(_info: &PanicInfo) -> ! {
    match pwm_ {
        Some(&mut pwm) => {
            pwm.disable();
        },
        None => {},
    };
}
#[entry]
fn main() -> ! {
let mut pwm = ...;
unsafe {
        pwm_ = Some(&mut pwm);
    }
pwm.set_duty(duty);
```

```
error[E0507]: cannot move out of `"*pwm_.0"` as `"pwm_"` is a static item
  --> src/main.rs:77:11
   |
77 |     match pwm_ {
   |           ^^^^ help: consider borrowing here: `&pwm_`
78 |         Some(&mut pwm) => {
   |                   ---
   |                   |
   |                   data moved here
   |                   move occurs because `pwm` has type `stm32f1xx_hal::pwm::Pwm<stm32f1::stm32f103::TIM3, stm32f1xx_hal::pwm::C1>`, which does not implement the `Copy` trait

error[E0596]: cannot borrow `pwm` as mutable, as it is not declared as mutable
  --> src/main.rs:79:13
   |
78 |         Some(&mut pwm) => {
   |                   --- help: consider changing this to be mutable: `mut pwm`
79 |             pwm.set_duty(0);
   |             ^^^ cannot borrow as mutable

error[E0596]: cannot borrow `pwm` as mutable, as it is not declared as mutable
  --> src/main.rs:80:13
   |
78 |         Some(&mut pwm) => {
   |                   --- help: consider changing this to be mutable: `mut pwm`
79 |             pwm.set_duty(0);
80 |             pwm.disable();
   |             ^^^ cannot borrow as mutable

error[E0597]: `pwm` does not live long enough
   --> src/main.rs:169:21
    |
89  | #[entry]
    |        - `pwm` dropped here while still borrowed
...
169 |         pwm_ = Some(&mut pwm);
    |         ------------^^^^^^^^-
    |         |           |
    |         |           borrowed value does not live long enough
    |         assignment requires that `pwm` is borrowed for `'static`

error[E0499]: cannot borrow `pwm` as mutable more than once at a time
   --> src/main.rs:247:9
    |
169 |         pwm_ = Some(&mut pwm);
    |         ---------------------
    |         |           |
    |         |           first mutable borrow occurs here
    |         assignment requires that `pwm` is borrowed for `'static`
...
247 |         pwm.set_duty(duty);
    |         ^^^ second mutable borrow occurs here
```

TheZoq2, 2019-10-20:
Try making the static mut the owner of the peripheral, rather than just having a mutable reference.

riaqn, 2019-10-23:
Thanks! it works.

Although I still hope to use reference since that would be simpler for me, but i'm closing this issue anyway.

closed 2019-10-23 by riaqn