stm32-rs/stm32f1xx-hal #127: help with driver for HTU21D temperature sensor

JPFrancoia, 2019-10-26:
Hi, I'd like to use a HTU21D humidity sensor with a blue pill. Unfortunately, this sensor doesn't have an "official" driver on this page: https://github.com/rust-embedded/awesome-embedded-rust/blob/master/README.md#driver-crates.

But I found *one* repository implementing the driver: https://github.com/etrombly/htu21d.
But I can't get it to work. Here is the code I'm using:

```rust
// #![deny(unsafe_code)]
// #![deny(warnings)]
#![no_main]
#![no_std]

extern crate cortex_m as cm;
extern crate cortex_m_rt as rt;
extern crate embedded_hal;
extern crate nb;
extern crate panic_itm;
extern crate stm32f1xx_hal as hal;
extern crate htu21d;
// extern crate itoa;

use nb::block;
use rt::entry;

// use embedded_hal::digital::v2::OutputPin;
use hal::prelude::*;
use hal::{
    i2c, pac,
    serial::{Config, Serial},
    delay::Delay,
};

use htu21d::Htu21df;

#[entry]
fn main() -> ! {
    // Initialize peripherals
    let device_peripherals = pac::Peripherals::take().unwrap();
    let cortex_peripherals = cortex_m::Peripherals::take().unwrap();

    let mut rcc = device_peripherals.RCC.constrain();
    let mut flash = device_peripherals.FLASH.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Initialize the GPIO A and B
    // let mut gpioa = device_peripherals.GPIOA.split(&mut rcc.apb2);
    let mut gpiob = device_peripherals.GPIOB.split(&mut rcc.apb2);

    // Prepare the alternate function I/O registers
    let mut afio = device_peripherals.AFIO.constrain(&mut rcc.apb2);

    let scl = gpiob.pb6.into_alternate_open_drain(&mut gpiob.crl);
    let sda = gpiob.pb7.into_alternate_open_drain(&mut gpiob.crl);

    let i2c = i2c::BlockingI2c::i2c1(
        device_peripherals.I2C1,
        (scl, sda),
        &mut afio.mapr,
        i2c::Mode::Standard { frequency: 400 },
        clocks,
        &mut rcc.apb1,
        50000,
        3,
        10000,
        50000,
    );

    let mut htu21d = Htu21df::new(i2c).unwrap();
    let humidity = htu21d.get_humidity().unwrap();

    loop {}
}
```

The sensor works (tested on an Arduino Uno), the blue pill + I2C works (I can read a bmp085 I2C sensor with more or less the same code).
But I can't seem to get anywhere with the htu21d on the blue pill, with the code above. While debugging, it seems the instruction `let mut htu21d = Htu21df::new(i2c).unwrap()` never completes.

From what I can tell from the debugger, I thinks the `reset` instruction triggered in the driver doesn't complete. It's stuck here: 

https://github.com/etrombly/htu21d/blob/693bcd73d7b618425ff767ae425f0b25011182c0/src/lib.rs#L38

Also, I tried disabling the `reset` to try another function from the driver (eg: `get_humidity`), but the code hangs as well.

For everything I tried, I never got an error, the code just seems to hang.

Do you have any idea where to start to find the problem?

Cheers

TheZoq2, 2019-10-26:
Hmm, strange. As far as I can tell, your setup code looks correct, allthough i'm not sure what `400` means. It could be hz, khz, mhz, or something else. We should probably fix that.

From reading the code, it looks like `reset` just calls `i2c::write` https://github.com/etrombly/htu21d/blob/master/src/lib.rs#L38 so something is *probably* wrong in the i2c code, or the driver does something which doesn't work with our hal for whatever reason.

Do you have access to a logic analyser, or something else that could see what data, if any goes back and forth?

JPFrancoia, 2019-10-26:
Thanks for your answer.

> Hmm, strange. As far as I can tell, your setup code looks correct, allthough i'm not sure what 400 means. It could be hz, khz, mhz, or something else. We should probably fix that.

Yes you're right about the frequency, the doc doesn't mention anything about the unit: https://docs.rs/stm32f1xx-hal/0.4.0/stm32f1xx_hal/i2c/enum.Mode.html#method.get_frequency. But it's a `u32`, so I assume I assume it's in Hz? I tried my code with `100_000` for the frequency, just in case, but that didn't solve the issue.

> Do you have access to a logic analyser, or something else that could see what data, if any goes back and forth?

I'm not sure what you mean by that. I use gdb-dashboard to debug normally.
The call to `write` that you mentioned goes into `write_without_stop`, then `send_start_and_wait`.

[image]

Then the code continues running until:

[image]

I tried to step into this call, but after a few steps, I ended up back at this line. So from what I can tell, this line makes the flow of the program wait.

Sorry if I'm not clear, I'm no debugger expert but let me know what I can do to help us make progress.

TheZoq2, 2019-10-27:
> Yes you're right about the frequency, the doc doesn't mention anything about the unit: https://docs.rs/stm32f1xx-hal/0.4.0/stm32f1xx_hal/i2c/enum.Mode.html#method.get_frequency. But it's a u32, so I assume it's in Hz? I tried my code with 100_000 for the frequency, just in case, but that didn't solve the issue.

Yep, it is most likely hz. In which case, 400 sounds very low.

> I'm not sure what you mean by that. I use gdb-dashboard to debug normally.

A device that can look at the signals being sent, kind of like an oscilloscope, but for digital signals.

> The call to write that you mentioned goes into write_without_stop, then send_start_and_wait.

IIRC i2c sends an addres, and waits for the receiver with the address to reply. It sounds to me like the device isn't responding for whatever reason

JPFrancoia, 2019-10-27:
> Yep, it is most likely hz. In which case, 400 sounds very low.

Ok, I now use `100_000` Hz, which is the standard frequency for I2C.

> A device that can look at the signals being sent, kind of like an oscilloscope, but for digital signals.

Sadly I don't have that :/ Can we do without it?

> IIRC i2c sends an address, and waits for the receiver with the address to reply. It sounds to me like the device isn't responding for whatever reason

yes you're right, I think the sensor isn't responding, and that's driving me crazy because I don't see why.

With the following code:

```rust
// #![deny(unsafe_code)]
// #![deny(warnings)]
#![no_main]
#![no_std]

extern crate cortex_m as cm;
extern crate cortex_m_rt as rt;
extern crate embedded_hal;
extern crate nb;
extern crate panic_itm;
extern crate stm32f1xx_hal as hal;
extern crate bmp085;
// extern crate itoa;

use nb::block;
use rt::entry;

// use embedded_hal::digital::v2::OutputPin;
use hal::prelude::*;
use hal::{
    i2c, pac,
    serial::{Config, Serial},
    delay::Delay,
};

#[entry]
fn main() -> ! {
    // Initialize peripherals
    let device_peripherals = pac::Peripherals::take().unwrap();
    let cortex_peripherals = cortex_m::Peripherals::take().unwrap();

    let mut rcc = device_peripherals.RCC.constrain();
    let mut flash = device_peripherals.FLASH.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Initialize the GPIO A and B
    let mut gpioa = device_peripherals.GPIOA.split(&mut rcc.apb2);
    let mut gpiob = device_peripherals.GPIOB.split(&mut rcc.apb2);

    // Prepare the alternate function I/O registers
    let mut afio = device_peripherals.AFIO.constrain(&mut rcc.apb2);

    let scl = gpiob.pb6.into_alternate_open_drain(&mut gpiob.crl);
    let sda = gpiob.pb7.into_alternate_open_drain(&mut gpiob.crl);

    let i2c = i2c::BlockingI2c::i2c1(
        device_peripherals.I2C1,
        (scl, sda),
        &mut afio.mapr,
        i2c::Mode::Standard { frequency: 100_000 },
        clocks,
        &mut rcc.apb1,
        50000,
        3,
        10000,
[... 78 lines cut ...]
        8000000
      ),
```

And here is the result of the call to `read`:
```
$2 = bmp085_driver::PT {
  temperature: 215,
  pressure: 101853
}
```

(temperature is indeed 21.5 in the room).

I compared the drivers for the bmp085 and the htu21d and they are very similar. They use the exact same functions. But the htu21d isn't responding.

- I checked the sensor (again) with Arduino this morning and it's working fine
- The I2C seems to work fine with another sensor, so I'd be inclined to say the problem doesn't come from hal. But what surprises me is that the `BlockingI2c` struct doesn't timeout. If the sensor isn't responding for whatever reason, I'd expect to get an error after a certain amount of attempts/time, since I passed this parameters when I created the I2C struct:

```rust
    start_timeout_us: u32,
    start_retries: u8,
    addr_timeout_us: u32,
    data_timeout_us: u32
```

TeXitoi, 2019-10-27:
For the timeout to work, you need to activate the DWT by memory.

Maybe you can try other modes.

TheZoq2, 2019-10-27:
> Sadly I don't have that :/ Can we do without it?
Of course, it would just have been a convenient way to check things.

Since the sensor works with an arduino, and the i2c works with another sensor, we can rule out a bunch of things. Off the top of my head, I can think of two things that may be going wrong:

1. Do you have pull-up resistors? I i2c needs them, and they may be integrated into the bmp085 but not the htu21d. 
2. The addresses are incorrect. This is less likely, but perhaps the htu21d is using a different default address than the arduino lib. This is only possible if the sensor supports changing the address, usually via some solder pad

JPFrancoia, 2019-10-27:
> For the timeout to work, you need to activate the DWT by memory.
> 
> Maybe you can try other modes.

So I changed the 3 first lines of my main to:

```rust
    // Initialize peripherals
    let device_peripherals = pac::Peripherals::take().unwrap();
    let mut cortex_peripherals = cortex_m::Peripherals::take().unwrap();

    cortex_peripherals.DWT.enable_cycle_counter();
```

To enable DWT. Is that how it should be done? Because it didn't change anything, the I2C doesn't time out (by the way, from a design point of view, it's counter-intuitive to do any extra step to enable the timing out while several time out parameters are passed to BlockingI2C at creation time).

I also tried the following modes/frequencies:
- Fast, frequency 100kHz, DutyCycle::Ratio2to1
- Fast, frequency 100kHz, DutyCycle::Ratio16to9
- Both ratios with frequency 400kHz

But no luck so far :/

> Do you have pull-up resistors? I i2c needs them, and they may be integrated into the bmp085 but not the htu21d.

I use the htu21d on a breakboard: https://learn.sparkfun.com/tutorials/htu21d-humidity-sensor-hookup-guide/all. On the page, it's clearly mentioned that pull-up resistors are included on the board. So yes, it seems I have pull-up resistors.
 
> The addresses are incorrect. This is less likely, but perhaps the htu21d is using a different default address than the arduino lib. This is only possible if the sensor supports changing the address, usually via some solder pad

Yes I thought about that as well, and I checked the library that I use on Arduino: https://github.com/sparkfun/SparkFun_HTU21D_Breakout_Arduino_Library/blob/master/src/SparkFunHTU21D.h

It uses the same address (`0x40`) for the reset. The other addresses seem correct as well. The difference with the Arduino lib is that they implement the blocking state themselves (they write to registers `NO_HOLD` and they explicitely wait/retry), while we use a BlockingI2c that probably does that.

JPFrancoia, 2019-10-30:
I'm closing this, I just got a BME280 sensor and it works perfectly by just using the proper driver.

I don't know what's happening with the HTU21D and the blue pill, but I'm pretty sure it's hardware related. My blue pill is a clone, maybe its specs aren't good enough for this sensor. Or maybe the sensor is slightly faulty, who knows :/

Thanks for your help.

TheZoq2, 2019-10-31:
Glad to hear you found some sort of a solution.

I doubt the specs are an issue, and I wouldn't be surprised if the HAL has some subtle bug. I've had similar issues with another sensor which I attributed to it being broken, but I don't think I ever tried it with a normal arduino. I'll re-open this and try to investigate this weekend

JPFrancoia, 2019-10-31:
No problem, thanks for the follow up. I'll keep watching this issue, let me know if you want me to test things.

TheZoq2, 2019-11-03:
Update: I tried the bno055 sensor I mentioned, and it doesn't work either. Though rather than freezing, it returns `Err(I2c(Acknowledge))`. I'll have to investigate further but it does look like there is an issue with the i2c implementation

TheZoq2, 2019-11-20:
I now have an update on this.

The acknowledge error happens after the `soft_reset` function is called in the driver, disabling it seems to fix the issue. I'm not sure if this is a driver problem, or a HAL problem, so I'll have to investigate further.

burrbull, 2019-12-07:
Is this related to #150 ?

eupn, 2020-01-11:
@TheZoq2 hey, did you used by bno055 driver? If you found a problem please file the issue/PR on bno055 repo.

TheZoq2, 2020-01-17:
@eupn I did, I'll see if I can still reproduce the issue, and then open a PR

TheZoq2, 2020-10-05:
I fixed my bmp055 issue, turns out it was a missing delay in the startup sequence. I'll close this for now since it seems like we haven't been able to find a HAL bug yet. Feel free to re-open if desired

closed 2020-10-05 by TheZoq2