stm32-rs/stm32f1xx-hal #129: Setting up a network of sensors: looking for BT or Wifi

JPFrancoia, 2019-10-30:
Hi,

I have several blue pills connected to temperature and humidity sensors. I would like to stick a bluetooth or wifi module to them so that I can connect them to a central node, let's say a Raspberry pi.

I have bluetooth modules and I'm almost sure I could make this work with bluetooth, but I'm not sure I'll be satisfied with the results since:
- BT isn't super reliable
- A master BT node can only be connected to 7 peripherals

So, I would like to know if there is a wifi module with an embedded-hal driver. Someting that works kinda out of the box and is reliable.

Do you know if there is one?

Also, I'm all ears if you have any suggestion on my network of sensors.

TheZoq2, 2019-10-30:
I'm sure there are a couple of options. I've used [these](https://www.letscontrolit.com/wiki/index.php/ESP-01) successfully and have a (work in progress) crate for controlling them at https://github.com/TheZoq2/esp01-rs

The esp01 crate is extracted from https://github.com/TheZoq2/weather and AFAIK doesn't fully work in the standalone form yet. I hope to fix that soon rather than later, but for now it'll have to just be inspiration :)

JPFrancoia, 2019-11-01:
Really cool :) I have a couple of questions:
- how far is your crate from being stable?
- what's your opinion of BT vs wifi for this use case?
- what do you mean by "doesn't fully work in the standalone form yet"? I don't necessarily to flash my code on the esp, I'm ok with controlling it from the blue pill

TheZoq2, 2019-11-01:
> how far is your crate from being stable?
For my use case, it has been stable enough, though it could probably use a bit more testing.

> what do you mean by "doesn't fully work in the standalone form yet"? I don't necessarily to flash my code on the esp, I'm ok with controlling it from the blue pill

By standalone form, I mean as a standalone crate. I extracted the code from the weather station repo because someone asked for it in discord. However, I didn't test the code, and realised that it's not useable yet (see the first issue in the repo). Hopefully I'll find some time to take care of that this weekend :)

> what's your opinion of BT vs wifi for this use case?
I haven't done much with BT so I can't comment much on that. As for wifi, the major hurdle I ran into with my project is how much power it consumes. It consumes about 200mA when in use and needs to do so for ~10 seconds every time you want to send data. Even with everything being asleep most of the time, and only waking every 10 minutes or so, power consumption turned out to be a problem for me. However, if you're not powering your things off a battery, there should be no problems.

Also, wifi is nice, because your hub can just be  normal web or TCP server which is easy to write.

JPFrancoia, 2019-11-01:
Thanks for your comments.

I don't intend to run anything on battery for this project so power consumption won't be a problem.
I'll give a chance to BT first in the next few days since I have the hardware already, and I'll try the esp after.

How many esp do you need to establish communication? Can I use only one for the sensor, and use the wifi of the raspberry pi? (just to be extra sure).

TheZoq2, 2019-11-01:
> How many esp do you need to establish communication? Can I use only one for the sensor, and use the wifi of the raspberry pi? (just to be extra sure).

Just one per sensor, and they can send data to any other device on the local network or internet, assuming they are connected to wifi, so a raspi is no issue. They can also act as web servers and wifi access points I believe, but that doesn't sound like what you need

JPFrancoia, 2019-11-01:
Thanks!
Yes I just want the sensors to send data, I'll put the server on the Pi.
I didn't dive through the code as of yet, but any chance the crate allows POST request on a Flask server? With encryption? (not much hope on the last one)

TheZoq2, 2019-11-01:
I *believe* POST support is in there or easy to implement. IIRC encryption is hard or impossible because of the limited amount of resources on the devices

TheZoq2, 2019-11-02:
Update: I pushed a new version of the crate which successfully compiles and should work fine. Let me know if you run into any issues :)

JPFrancoia, 2019-11-05:
Perfect thanks! I just ordered 3 ESP, I'll have a go at it when I receive them. I'll keep you posted :)

TheZoq2, 2019-11-08:
Nice, let me know if you find any issues with them! I'll close this issue since there isn't anything actionable here :)

closed 2019-11-08 by TheZoq2