stm32-rs/stm32f1xx-hal #138: Innaccurate Serial Example in 0.4.0 on docs.rs

IGBC, 2019-11-18:
Hi, 

The example for the USART on docs.rs is out of date for version 0.4.0. 

https://docs.rs/stm32f1xx-hal/0.4.0/stm32f1xx_hal/serial/index.html

I went to fix it and found its already been corrected on master. 

Someone I think just needs to upload the patch to docs.rs.

Thanks.

TheZoq2, 2019-11-19:
Yep, the docs version tends to drift away from the github version, pretty quickly.

We should probably do a release soon though.

IGBC, 2019-11-19:
I'm not sure a release helps in my case.
 
I know everything around here is still very alpha, but we've bound our project to a given version (0.4.0) so those docs will stay broken for us.
 
To even be able to consider updating we'd need version to version changelogs.

TheZoq2, 2019-11-19:
As far as I know, there is no way to update a crate on docs.rs or crates.io. We could release a version 0.4.1 to fix the issue though.

We do have version to version changes in `CHANGELOG.md`.

Or do you want changelogs for the examples?

IGBC, 2019-12-03:
no point if you're releasing 0.5.0 soon

closed 2019-12-03 by IGBC

TheZoq2, 2019-12-03:
Good timing, 0.5.0 is released :)