stm32-rs/stm32f1xx-hal #143: Not building with features stm32f100 or stm32f101

pdgilbert, 2019-11-25:
(I'm a newbie so this may be a simple error or misunderstanding on my part, but I think it is a bug or not yet implemented feature.)

I have a blink example, and several others, that build using stm32f1xx-hal with `--feature stm32f103`but with `--features stm32f100` I get
```
 = note: rust-lld: error: undefined symbol: TAMPER

          >>> referenced by stm32f1.4gk9g0mv-cgu.0
 ...
 rust-lld: error: undefined symbol: TIM6
```

and with -`-features stm32f101` I get
```
= note: rust-lld: error: undefined symbol: ADC1_2

          >>> referenced by stm32f1.4gk9g0mv-cgu.0
  ...
  rust-lld: error: undefined symbol: DMA2_CHANNEL3
```
The code is in examples/ at https://github.com/pdgilbert/eg_stm_hal and the build details are in Travis CI under https://travis-ci.org/pdgilbert/eg_stm_hal

TheZoq2, 2019-11-25:
That's odd. `cargo build --features stm32f100 --example blinky`, and the same thing with `101` works for me.

What version of the crate are you building? Git master or the crates version?

Does this happen locally, or only on travis?

pdgilbert, 2019-11-25:
This is a `blink` example in my code which uses stm32f1xx-hal, not the `blinky` example in the stm32f1xx-hal crate. I think the main difference is that I'm using different pins and blinking 3 leds, not the one on board.   I'm using git master. It happens both locally and on travis. All the example build details are on Travis at https://travis-ci.org/pdgilbert/eg_stm_hal .

pdgilbert, 2019-11-25:
(Other than some warnings, `cargo build --features stm32f100 --example blinky` in my stm32f1xx-hal checkout does build, as does `101`. The problem is the differences in my blink code, which works with `103` but does not build with `100` or `101`.)

TeXitoi, 2019-11-25:
I suspect that the features are not working as expected in your project, but I can't say more as I'm not an expect in cargo features.

pdgilbert, 2019-12-02:
This seems to have been happening because I specified the features incorrectly in my package. I had specified them the same way as in  stm32f1xx-hal's  `Cargo.toml` but in my package they needed to be specified to indicate I way passing them to `stm32f1xx-hal`. It seems the stm32f103 feature setting worked only accidentally, I think because it was first.  For the record, what works is
```
[features]
stm32f100 = ["stm32f1xx-hal/stm32f100", "stm32f1xx-hal/rt"]
stm32f101 = ["stm32f1xx-hal/stm32f101", "stm32f1xx-hal/rt"]
stm32f103 = ["stm32f1xx-hal/stm32f103", "stm32f1xx-hal/rt"]
stm32f303 = ["stm32f3xx-hal/stm32f303", "stm32f3xx-hal/rt"] 
```
It seems also necessary to specify 
```
[dependencies.stm32f1xx-hal]
optional = true

[dependencies.stm32f3xx-hal]
optional = true
```
to avoid having the two hal crates interfere with one another.

closed 2019-12-02 by pdgilbert