stm32-rs/stm32f1xx-hal #144: pwm example compile error

burrbull, 2019-11-25:
```
error[E0599]: no function or associated item named `tim4` found for type `stm32f1xx_hal::timer::Timer<_>` in the current scope
  --> examples/pwm.rs:49:26
   |
49 |     let mut pwm = Timer::tim4(p.TIM4, &clocks, &mut rcc.apb1)
   |                          ^^^^
   |                          |
   |                          function or associated item not found in `stm32f1xx_hal::timer::Timer<_>`
   |                          help: there is a method with a similar name: `tim1`

error: aborting due to previous error
```
on f103

#133 ?

burrbull, 2019-11-26:
Example should be run with ` --features="stm32f103, medium"`

TheZoq2, 2019-11-26:
This should probably be mentioned in the comments for the example. I'll re-open this and hopefully fix it tonight

PR #148 (2019-12-03): Documentation updates and cleanup
This fixes #144, both in terms of making that example compile without extra features, and documenting the features.

While i'm at it, I also updated the PWM and timer docs, and cleaned up `Cargo.toml`
closed 2019-12-03 by TheZoq2