stm32-rs/stm32f1xx-hal #154: Docs.rs builds failed

jamwaffles, 2019-12-17:
The 0.5.1 and 0.5.2 docs builds are failing with an error about not being able to build for `thumbv7m-none-eabi`. Example log [here](https://docs.rs/crate/stm32f1xx-hal/0.5.2/builds/205703). 0.5.0 is fine.

I attempted a fix locally by adding `default-target` to `Cargo.toml`:

```diff
  [package.metadata.docs.rs]
  features = ["stm32f103", "rt", "stm32-usbd"]
+ default-target = "thumbv7m-none-eabi"
```

then following the [docs.rs instructions](https://github.com/rust-lang/docs.rs) and running 

```bash
docker-compose run -v "$(realpath ~/Repositories/stm32f1xx-hal)":/build web build crate --local /build
```

But I get the same error as the log on docs.rs:

```
2019/12/17 18:30:06 [INFO] rustwide::cmd: running `"docker" "start" "-a" "0fc8e7382b5a4976d92d59f972cd57b6c50dfdcfaacccc41044654148e92838d"`
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling typenum v1.10.0
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling semver-parser v0.7.0
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling proc-macro2 v0.4.30
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling unicode-xid v0.1.0
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]     Checking stable_deref_trait v1.1.1
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling cortex-m v0.6.1
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]     Checking vcell v0.1.2
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling syn v0.15.34
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling rand_core v0.4.0
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling cortex-m v0.5.10
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]    Compiling cortex-m-rt v0.6.11
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]     Checking r0 v0.2.2
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error[E0463]: can't find crate for `core`
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]   |
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]   = note: the `thumbv7m-none-eabi` target may not be installed
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error[E0463]: can't find crate for `core`
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]   |
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]   = note: the `thumbv7m-none-eabi` target may not be installed
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error: aborting due to previous error
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error: aborting due to previous error
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] For more information about this error, try `rustc --explain E0463`.
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] For more information about this error, try `rustc --explain E0463`.
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error[E0463]: can't find crate for `core`
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]   |
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]   = note: the `thumbv7m-none-eabi` target may not be installed
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error: aborting due to previous error
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr]
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error: could not compile `vcell`.
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] warning: build failed, waiting for other jobs to finish...
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] For more information about this error, try `rustc --explain E0463`.
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error: could not compile `r0`.
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] warning: build failed, waiting for other jobs to finish...
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] error: could not compile `stable_deref_trait`.
2019/12/17 18:30:06 [INFO] rustwide::cmd: [stderr] warning: build failed, waiting for other jobs to finish...
2019/12/17 18:30:07 [INFO] rustwide::cmd: [stderr] error: build failed
```

This might be more relevant to the docs.rs repo, but I thought I'd open it here in case it's a fixable problem on the stm32-rs side of things.

TheZoq2, 2019-12-19:
I noticed that as well but forgot to deal with it. I'm not sure how to fix it outside of reporting it to docs.rs

jamwaffles, 2019-12-19:
That's fair. I'll open an issue on the docs.rs repo when I have a minute.

dfrankland, 2019-12-29:
We're having the same issue over on `stm32f3xx-hal`. Until `thumbv7m-none-eabi` is added to doc.rs, are there any issues with making `default-target = "x86_64-unknown-linux-gnu"`?

TheZoq2, 2020-01-03:
That would be a solution, is there a way to do that for just the docs build?

dfrankland, 2020-01-03:
That's what I am proposing to do, but I am not sure if it works yet though. I assume something like this should:
```diff
  [package.metadata.docs.rs]
  features = ["stm32f103", "rt", "stm32-usbd"]
+ default-target = "x86_64-unknown-linux-gnu"
```

PR #172 (2020-01-19): Improve quick start docs
I recently helped a friend start using this and noticed some documentation problems which this hopefully fixes:

- Programs to install wasn't documented
- Having to specify target when compiling
- linker, gdb and cargo config wasn't documented
- inline example was out of date (replaced with a link to the example)

I also decided to remove some redundant text from lib.rs and link to README instead, just to avoid having some parts of the docs drift out of date.

Oh, and I fixed the docs.rs build, so this closes #154 

The docs are very linux centric at the moment, but I  [... cut]
closed 2020-01-19 by therealprof