stm32-rs/stm32f1xx-hal #162: Push new release to crates.io?

fuchsnj, 2020-01-06:
I just spent some time cloning this repo and making a change, only to find out it was already implemented (https://github.com/stm32-rs/stm32f1xx-hal/pull/160). Partially my fault, I didn't find an open issue for it so I started working on it.

Is there any reason that change hasn't been released on crates.io?

I also noticed crates.io has a later version (0.5.2) than master (0.5.1) in the Cargo.toml. (It was updated in the README though).

TheZoq2, 2020-01-10:
Hi.

There is no particular reason for it not being released, I just haven't gotten around to it. I'll do it tonight if I remember.

And the 0.5.2/0.5.1 issue is probably me forgetting to run `git push upstram` after doing the release. Also, 0.5.2 is just 0.5.1 with some minor doc fixes because I messed up the release :wink:

TheZoq2, 2020-01-20:
`v0.5.3` has been released :tada:

closed 2020-01-20 by TheZoq2