stm32-rs/stm32f1xx-hal #166: Implement PWM Trait

jounathaen, 2020-01-14:
For me as a stupid novice it looks like this crate does not implement the Pwm trait from the embedded-hal.
https://docs.rs/embedded-hal/0.2.3/embedded_hal/trait.Pwm.html

However, it seems to me, as if only the `set_period` and `get_period` functions are missing.

What a pity that these are the functions I need right now :pensive:

TheZoq2, 2020-01-15:
Yep, you are correct in that assessment. If I recall correctly, the reason that is the case is that the period is shared among all channels, so period changes would affect more than just a single pin.

Off the top of my head, I'm not sure what the best way to add the functionality would be, but if you're interested I'd gladly try to point you in the right direction for where to look to fix this.

justacec, 2020-01-21:
I was having a similar issue as the OP.  After some discussions with @TheZoq2, I have put forward PR #176 and have requested code review.  (This was both more fun and frustrating than I thought it would be :) )

tib888, 2020-04-11:
> Yep, you are correct in that assessment. If I recall correctly, the reason that is the case is that the period is shared among all channels, so period changes would affect more than just a single [...]

I have just checked the embedded_hal version = "0.2.3", and it seems perfectly fine: there is no period setter in the PwmPin. The only difference is that every function name starts with a try_, and try return a result. so it could be implemented based on the current implementation...

TheZoq2, 2020-10-05:
I think this is fixed now, feel free to re-open if I'm missremembering

closed 2020-10-05 by TheZoq2