stm32-rs/stm32f1xx-hal #167: Impl Read<u8> / Write<u8> for Serial

dbrgn, 2020-01-17:
Is there a reason why Read<u8> / Write<u8> is only implemented for `Rx` and `Tx` (obtainable via `split()`) but not on the `Serial` type directly?

Other HALs like the [stm32f4xx-hal](https://docs.rs/stm32f4xx-hal/0.6.0/stm32f4xx_hal/serial/struct.Serial.html) have such impls.

I'm working with a driver that wants a `Read<u8> + Write<u8>` type and I don't want to create a wrapper type with my own impl if not necessary :slightly_smiling_face:

TheZoq2, 2020-01-17:
I don't believe there is an underlying reason, it's just that nobody has gotten around do doing it yet  :)

dbrgn, 2020-01-17:
Hm, I tried to implement it, but I realized that for this struct:

```
pub struct Serial<USART, PINS> {
    usart: USART,
    pins: PINS,
}
```

...the Read / Write traits cannot be implemented, as it is not known how to get to the pin types from the generic `PINS` type argument.

One workaround would be to implement `Read / Write` for a tuple `(Tx, Rx)`, but that's a bit of a hack in my opinion.

The better solution in my eyes would be to replace the `Pins` trait with separate `TxPin` / `RxPin` traits, as I did in the [stm32l0xx-hal](https://github.com/stm32-rs/stm32l0xx-hal/pull/68). Any opinions on that? It would be a breaking change, but we're still on version 0.x so I guess that's fine :slightly_smiling_face:

TheZoq2, 2020-01-17:
> Any opinions on that? It would be a breaking change, but we're still on version 0.x so I guess that's fine slightly_smiling_face

Sounds like a good idea. We have breaking changes all the time :wink:

PR #171 (2020-01-18): Impl Read<u8> / Write<u8> for Serial
Fixes #167.

I managed to implement this without having split tx/rx pins in the `Serial` struct thanks to the help of @Disasm. Instead, it uses a private trait (`UsartReadWrite`) to avoid having duplicated peripheral access logic. There should be no breaking change in this PR.
closed 2020-01-18 by TheZoq2