stm32-rs/stm32f1xx-hal #169: Delay timing is consistently ~5 microseconds slow

fuchsnj, 2020-01-17:
I'm debugging some issues in a 1-wire implementation (which is very timing sensitive) and I noticed that `DelayUs` seems to always take about ~5.25 microseconds longer than requested.

For example, this program
```rust
fn run_loop<P, E>(
    delay: &mut (impl DelayUs<u16> + DelayMs<u16>),
    mut tx: impl Write,
    mut one_wire_bus: OneWire<P>,
    debug: &mut (impl OutputPin),
) -> !
    where
        P: OutputPin<Error=E> + InputPin<Error=E>,
        E: Debug
{
    loop {
        debug.set_high();
        delay.delay_us(1);
        debug.set_low();

        delay.delay_ms(1);
    }
}
```
Should set the pin high for 1 microsecond, then turn it off. It's actually taking ~6.25 micros.
[image]

I've tested without the delay, and the pin can turn on/off in significantly less than 1 microsecond, so I'm pretty confident this is the delay being too slow. I've also reproduced at larger delays.

The onewire protocol requires accurate delays as low as around 6 micro-seconds, so this is nearly doubling the delay and causing significant errors.

TeXitoi, 2020-01-17:
Like that, we can't see which implementation of delay you are using, and thus we can't know which one to check.

Are you running your code in release mode? If not, that's should be the overhead of function call and all the small things that are not optimized out in debug.

fuchsnj, 2020-01-17:
I am using the `STM32F103C8T6`
```
[dependencies.stm32f1xx-hal]
version = "0.5.1"
features = ["stm32f103", "rt"]
```

I am creating the Delay with 
```rust
let dp = pac::Peripherals::take().unwrap();
let mut flash = dp.FLASH.constrain();
let mut rcc = dp.RCC.constrain();
let clocks = rcc.cfgr.freeze(&mut flash.acr);
let cp = cortex_m::Peripherals::take().unwrap();
let mut delay = Delay::new(cp.SYST, clocks);
```

and the debug pin with...
```rust
...
let mut gpiob = dp.GPIOB.split(&mut rcc.apb2);
let mut debug_pin = gpiob.pb13.into_push_pull_output(&mut gpiob.crh);
```

I am running in release mode with the following settings
```toml
[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
```
I've also tried `opt-level = 2` with the same results.

TheZoq2, 2020-01-17:
That does seem strange. Release mode was my  first guess as well, but I guess that's not the problem. I'lll try to look into it this weekend.

Is the incorrectness constant, i.e. is it always `delay+5us` or does it change depending on the delay you use?

geomatsi, 2020-01-17:
Hello @fuchsnj 

Maybe the reason is in the clock settings that you are using ? Your code snippet does not show any specific clock configuration. So you might be using default values. I don't remember default clock settings but they should be fairly conservative since a board may not have an external quartz. If you are using STM32F103x board with external quartz (e.g. BluePill), then you could try to enable HSE and to increase your clock settings, e.g. something like the following:

```
let clocks = rcc.cfgr.use_hse(8.mhz()).sysclk(32.mhz()).pclk1(16.mhz()).freeze(&mut flash.acr);
```

Regards,
Sergey

thalesfragoso, 2020-01-17:
Like @geomatsi said, I think the problem might be the clock.
The default clock is 8Mhz, which gives you 8 cycles per microsecond, this is quite tight for those requirements. If you check https://github.com/stm32-rs/stm32f1xx-hal/blob/68d639c716bae837035e5a1879a60717cda070c2/src/delay.rs#L48 you see that there are some calculations and registers IO to correctly set up the delay, which can quite easily exceed 8 cycles.

I can think of some solutions:

- **Crank up the clock:** Probably the best one and I would recommend doing it even if you combine with other solutions, if you're dealing with microsecond precision 8Mhz seems quite slow, the F1 goes up to 72Mhz with external crystal and 64Mhz with internal, there are some flash wait stages at this clock, but still can be way better than 8Mhz.

- **Play with periodic timers and interrupts:** You can also use peripherals timers for this, but like the delay thing, there are calculations before the start, but only the first time you start it, after that it restarts automatically but this only works with the same delay.

- ***Nop* your way out:** You might want to pick this one, it's kinda sad to take the whole path to configure a timer to just wait 1-10us, but using this you have to make sure to not get interrupted if you really need those greedy microseconds of precision. See https://docs.rs/cortex-m/0.6.2/cortex_m/asm/fn.delay.html and https://crates.io/crates/asm-delay.

Another solution is to do the register fiddling yourself and don't use the APIs, this way you can set up the thing before when you have the time, so you only need to start the peripheral when the moment comes, but to be honest, if you can, just go with the *nop* solution.

fuchsnj, 2020-01-18:
Thanks @geomatsi I think this was the main issue. I'm still pretty new to embedded development, and coming from the Arduino world, I naively assumed that clocks would default to the "highest" value. After increasing the clocks, the results are close enough to what I expect that it shouldn't be an issue anymore.

Thanks everyone for helping out. I'm impressed with how many quick responses I got!

Just for fun, I played around with a few settings. (@TheZoq2 this should answer your question too)

| Config  | 1 μs delay time |10 μs delay time |100 μs delay time |
| ------------- | ------------- | ------------- | ------------- |
| debug / default clock  | ~115 μs  | 123 μs | 208 μs |
| debug / 72 mhz  | 16 μs  | 25.8 μs | 115.2 μs |
| release / default clock  | 6.25 μs  |15.4 μs | 105.4 μs |
| release / 32 mhz  | 2.5 μs  | 11.4 μs| 101.5 μs |
| release / 72 mhz  | 1.65 μs  |10.6 μs | 100.7 μs |

fuchsnj, 2020-01-18:
I've confirmed this fixed my issue. I can reliably communicate with devices over a 1-wire bus now. Thanks again!

closed 2020-01-18 by fuchsnj