stm32-rs/stm32f1xx-hal #184: Hardware tests

arrowcircle, 2020-02-11:
Hi! I was thinking about unit tests and hardware.
It looks like a good idea to have number of tests that cover hal for most popular boards, like bluepill or discovery.

Is there any way to easily run unit test on the hardware? What hardware is needed in general to execute "full" suit of tests except test board?

TheZoq2, 2020-02-12:
This is something I've also been thinking about, but I have no idea how or if it is usually done. I have considered having one microcontroller or FPGA which would read the pins on the controller being tested, but I such a project would take quite a bit of effort I believe. Also, I'm not sure how one would program it

arrowcircle, 2020-02-12:
@TheZoq2 Have you seen platformio approach for remote tests?

I think it should not read pins, but check actual logic results in code. Or have a possibility to do tests with pins or with logic. If code is fine and hal is separated from logic, unit tests are easy to make.

But how to run them on hardware? system should have wrapper, that will prepare state of the MCU, run the test and store the result on the tests worker. Does this logic make sense?

pdgilbert, 2020-06-15:
(I'm just a newbie trying to learn about Rust and embedded programming.) Having worked some examples I have thought a few times that some sort of hardware unit testing would be pretty neat, but realized I am a long way from anything like that. If anyone figures it out I would really like to hear. However, I do see that there is still a need for some very doable CI testing for Rust stm32**-hal's. The code for hals seems very much in development mode, and some hals seem out-of-date or not in sync. So, I've tried to put together some simple examples in a repository with CI testing for the builds, and very manual less frequent checks to see if the examples actually run correctly on hardware. I would certainly appreciate comments and ideas from people with more experience than I have. A summary is at https://github.com/pdgilbert/eg_stm_hal#summary-of-examples. Scroll down a bit to see what works and doesn't. The CI results are at https://travis-ci.org/pdgilbert/eg_stm_hal. Expanding the CI for other hal's is pretty simple, I can do that if anyone would like. Actually checking that examples run will require hardware, so if you do that please leave details/results as an issue on the repository site.