stm32-rs/stm32f1xx-hal #197: Where does crate::pac live?

smedellin90, 2020-04-07:
I was looking at how the rtc is initialized for this stm32 family, and I noticed the following:
https://github.com/stm32-rs/stm32f1xx-hal/blob/37d1a7b3ec7d8a7edf5e1a178d8fd2863a4a578f/src/rtc.rs#L15

I'm new to rust, but I know that `use crate` refers to the current crate. But when I look for the module **pac** in the repository, it is nowhere to be found. Where is the source code for pac?

thalesfragoso, 2020-04-07:
It's a re-export with alias, see here:
https://github.com/stm32-rs/stm32f1xx-hal/blob/37d1a7b3ec7d8a7edf5e1a178d8fd2863a4a578f/src/lib.rs#L83-L99

smedellin90, 2020-04-07:
I see! Thank you for the explanation, @thalesfragoso

smedellin90, 2020-04-07:
Are the pac module and stm32 module the same? I'm looking through the stm32f3xx-hal repository, and they have have re-export aliased in a similar way, but instead of pac, they name the module stm32. 

Below is a screenshot of docs.rs for stm32f1xx-hal:

[image]

thalesfragoso, 2020-04-08:
There are several alias, in this repo the `pac` is the preferred one, but others are kept for retro compatibility.

smedellin90, 2020-04-08:
Awesome, thanks again!

closed 2020-04-08 by smedellin90