stm32-rs/stm32f1xx-hal #199: Pwm implementation is missing for Alternate<OpenDrain>

tib888, 2020-04-11:
Currently the Pwm implementation works only for Alternate<PushPull>.
I miss the implementation for Alternate<OpenDrain>, furthermore I was not able to figure out how to fix this.
Perhaps this Alternate abstraction should become independent of outputmode, and constraints should be changed where both are needed.

TheZoq2, 2020-04-13:
Hmm, from a quick skim of the datasheet, I don't see a reason for it being restricted to OpenDrain.

I believe the place where it gets set to `PushPull` is here: https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/pwm.rs#L111. Changing that `PushPull` to a generic parameter *should* fix the issue if I'm not misstaken

Ranlvor, 2023-08-12:
I think this issue can be closed. I am using PWM in OpenDrain-Mode and it works fine.

 #358 even says that is fixed this issue.

closed 2023-08-12 by burrbull