stm32-rs/stm32f1xx-hal #200: RX DMA Access on the SPI Peripheral

Fotte, 2020-04-13:
Hi there,

I am fairly new to rust, but i have done some embedded development in the past. I could not find a forum or something similar to post my question - propably github issues are not the right way - if thats the case it would be nice if you could point me in the right direction.

I am currently trying to access an SPI ADC using DMA. I want to trigger my DMA on the reception of an external interrupt. Using the DMA i need to send and receive 5 bytes [CMD][RES0 ... RES3].

The processor side is clear to me but right now i have no clue how this could be done using this library. Could you help me out?

TheZoq2, 2020-04-13:
I believe we have only implemented SPI tx over DMA but there is no reason it couldn't be implemented for rx as well IIRC. If you would like to attempt to implement it, I'll gladly provide some pointers for where to start etc.

Fotte, 2020-04-13:
Wow, never expected such a fast answer. Yeah i would like to give it a shot in the next days. 

My main concern right now is understanding the intention behind the DMA implementation. Is there anything (Mailinglist entrys) that i could read? I'm comming from a C background so there are many rust principles that are new to me.

therealprof, 2020-04-13:
@Fotte You might want check out the Rust embedded matrix channel, https://matrix.to/#/#rust-embedded:matrix.org . There's lot's of ongoing discussion about how to implement DMA use cases properly there. Although I think a blocking implementation should be rather trivial (though potentially also not that interesting).

TheZoq2, 2020-04-14:
> Is there anything (Mailinglist entrys) that i could read?

The embedonomicon has a section on implementing DMA. https://docs.rust-embedded.org/embedonomicon/dma.html. You can also get some inspiration from the current implementations for UART, ADC and SPI TX, and the relevant PRs #104 #99, #55 and #11

Windfisch, 2021-05-14:
I worked on this in the last days and will file a PR tomorrow or so; currently, I both have implementations for receiving only, and for receiving-while-transmitting. The latter is usable, while the former does not do anything useful in SPI master mode, because without transmitting, there is no clock and without a clock, there are no bits received.
Do you think the RX-only mode is still useful (e.g. in slave mode)? I never used slave mode so far.