stm32-rs/stm32f1xx-hal #212: Support for ADC multichannel continuous conversion?

yjh0502, 2020-05-02:
Hi, I'm trying to run a ADC multichannel (scan mode) continuous conversion. I dig into the code to check current status:

`Continuous` conversion: set `cont()` bit on `start()`
https://github.com/stm32-rs/stm32f1xx-hal/blob/89600ec0708cc9cb66c72c1b2c4b6b3a4827c7cc/src/adc.rs#L558-L568

`Scan` conversion: not setting `cont()` bit on `start()` clear `cont()` explicitly on initialization. It is definitely not intended for continuous conversion.
https://github.com/stm32-rs/stm32f1xx-hal/blob/89600ec0708cc9cb66c72c1b2c4b6b3a4827c7cc/src/adc.rs#L608-L617

I think there are several ways to support continuous scan conversion mode:

 - I could add quick and dirty fix by introducing `ContinuousScan` struct (or `struct Scan { continuous: bool }` if it's okay to break API), but I'm not sure if it's a right way.
 - I could think of generalized struct/trait which covers single/multichannel one-shot/continuous conversion scenario. Single-channel conversion could be treated as a special case of multichannel conversion anyway.

TheZoq2, 2020-05-02:
I'm not quite sure what you're asking, as far as I remember, the DMA should do continuous conversion, or does it just do multiple conversions on the same channel?

I think breaking the API is perfectly fine, we do that quite often.

yjh0502, 2020-05-02:
https://www.st.com/resource/en/application_note/cd00258017-stm32s-adc-modes-and-their-applications-stmicroelectronics.pdf

[image]

I mean `Scan` mode on `stm32f1xx-hal` API with continuous conversion.

On current API, we should choose between
 - single-channel single-shot conversion (`OneShot::read` API)
 - single-channel continuous conversion with DMA (`Adc::with_dma`, which returns `Continuous` mode) or,
 - multi-channel single-shot conversion with DMA (`Adc::with_scan_dma`, which returns `Scan` mode, scanning multiple channels by order, but only once)

We could run multi-channel single-shot conversion with DMA by `Adc::with_scan_dma` API, but it does not support continuous conversion. For example, with given scan sequence `CH0 -> CH1 -> CH2 -> CH3` and continuous mode (with `cont()` bit), it will run `CH0 -> CH1 -> CH2 -> CH3 -> CH0  -> CH1 -> ...` indefinitely and continuously.

I'll try to prepare a POC PR :)