stm32-rs/stm32f1xx-hal #216: How to reset DMA channel?

e-max, 2020-05-16:
Hi guys,
I am sorry if this is not an appropriate place to ask such a question, but I couldn't think of another one.  

I am trying to use the USART Rx device with DMA channel and it works great but only once.
How could I restart the channel after first use? 

The simplest example of the code.

```
        let pb10 = gpiob.pb10.into_alternate_push_pull(&mut gpiob.crh);
        let pb11 = gpiob.pb11;

        let channels = device.DMA1.split(&mut rcc.ahb);
.
        let serial = Serial::usart3(
            device.USART3,
            (pb10, pb11),
            &mut afio.mapr,
            Config::default().baudrate(115200.bps()),
            clocks,
            &mut rcc.apb1,
        );

        let (_, rx) = serial.split();

        let rx = rx.with_dma(channels.3);

        let buf = cortex_m::singleton!(: [u8; DMA_BUF_SIZE] = [0; DMA_BUF_SIZE]).unwrap();

        let transfer: Transfer<dma::W, &'static mut [u8; DMA_BUF_SIZE], RxDma3> = rx.read(buf);
        let (b, r) = transfer.wait();
        hprintln!("b = {:?}", b).unwrap();
// Works great until now, but whan I try to receive next batch.

        let transfer: Transfer<dma::W, &'static mut [u8; DMA_BUF_SIZE], RxDma3> = r.read(b);
        let (b2, _r2) = transfer.wait();  
//it's never return from this wait() no matter how much data I send through USART/
        hprintln!("b2 = {:?}", b2).unwrap();

```

e-max, 2020-05-18:
I rewrote my code using directly channel like this
```
        let pb10 = gpiob.pb10.into_alternate_push_pull(&mut gpiob.crh);
        let pb11 = gpiob.pb11;

        let channels = device.DMA1.split(&mut rcc.ahb);

        let serial = Serial::usart3(
            device.USART3,
            (pb10, pb11),
            &mut afio.mapr,
            Config::default().baudrate(115200.bps()),
            clocks,
            &mut rcc.apb1,
        );

        let (_, _rx) = serial.split();
        let buf = cortex_m::singleton!(: [u8; DMA_BUF_SIZE] = [0; DMA_BUF_SIZE]).unwrap();

        let mut channel = channels.3;

        {
            let buffer = as_slice::AsMutSlice::as_mut_slice(buf);
            channel
                .set_peripheral_address(unsafe { &(*USART3::ptr()).dr as *const _ as u32 }, false);
            channel.set_memory_address(buffer.as_ptr() as u32, true);
            channel.set_transfer_length(buffer.len());
        }
        core::sync::atomic::compiler_fence(core::sync::atomic::Ordering::Release);
        channel.ch().cr.modify(|_, w| {
            w.mem2mem()
                .clear_bit()
                .pl()
                .medium()
                .msize()
                .bits8()
                .psize()
                .bits8()
                .circ()
                .clear_bit()
                .dir()
                .clear_bit()
        });
        channel.start();

        while channel.in_progress() {
            hprintln!("buf {:?}", buf).unwrap();
            delay.delay_ms(1000 as u16);
        }
        hprintln!("final buf {:?}", buf).unwrap();
        for i in 0..8 {
            buf[i] = 0;
        }

        channel.stop();

        {
            let buffer = as_slice::AsMutSlice::as_mut_slice(buf);
            channel
                .set_peripheral_address(unsafe { &(*USART3::ptr()).dr as *const _ as u32 }, false);
            channel.set_memory_address(buffer.as_ptr() as u32, true);
            channel.set_transfer_length(buffer.len());
        }
        core::sync::atomic::compiler_fence(core::sync::atomic::Ordering::Release);
        channel.ch().cr.modify(|_, w| {
            w.mem2mem()
                .clear_bit()
                .pl()
                .medium()
                .msize()
                .bits8()
                .psize()
                .bits8()
                .circ()
                .clear_bit()
                .dir()
                .clear_bit()
        });
        channel.start();

        while channel.in_progress() {
            hprintln!("buf {:?}", buf).unwrap();
            delay.delay_ms(1000 as u16);
        }
        hprintln!("final buf {:?}", buf).unwrap();
```
and it still shows the same weird behavior - the first run finishes successfully, but during the second run I get only one byte in the buffer and that is all.

e-max, 2020-05-18:
it seems that my code was just too slow and I got an ORE  overflow between the moment I stopped an old DMA channel and started a new one.   I am sorry for bothering you.

closed 2020-05-18 by e-max

thalesfragoso, 2020-05-18:
@e-max That is no bother at all. I'm glad you managed to found the problem. I actually want to thank you for taking the time to create and update this issue, it will most definitely help someone in the future.

Also, I think we should probably do something about this, I'm not sure if we should clear this before starting the DMA, but we should at least provide a method to the user to clear this flag in the uart abstraction.