stm32-rs/stm32f1xx-hal #227: Get error trying to build cargo

yehielliv, 2020-06-15:
I'm getting this error at cargo build
error: Target not found. A `--features <target-name>` is required.
  --> C:\Users\KNC\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32f1xx-hal-0.6.0\src\lib.rs:80:1
   |
80 | compile_error!("Target not found. A `--features <target-name>` is required.");

This is my cargo.toml code
[package]
name = "stm32-blue-pill"
version = "0.1.0"
authors = ["yehielliv <386653+yehielliv@users.noreply.github.com>"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
embedded-hal = "0.2.3"
nb = "0.1.2"
cortex-m = "0.6.0"
cortex-m-rt = "0.6.8"
# Panic behaviour, see https://crates.io/keywords/panic-impl for alternatives
panic-halt = "0.2.0"
panic-semihosting = "0.5.2"
cortex-m-semihosting = "0.3.3"
stm32f1xx-hal = "0.6.0"

Any idea why?

Thanks.

TeXitoi, 2020-06-16:
You need to specify the specific MCU using a feature

https://github.com/TeXitoi/blue-pill-quickstart/blob/master/Cargo.toml#L8

yehielliv, 2020-06-16:
I did what you suggested exactly but now I get error
error[E0463]: can't find crate for `panic_semihosting`
 --> src\main.rs:5:1
  |
5 | extern crate panic_semihosting;
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate

pdgilbert, 2020-06-17:
It is a bit unclear if your cargo.toml has uncommented comments like "See more keys..."  I'm not sure what that would do. Maybe try putting  `#`  in front of those lines?  You might also try `cargo update`.

TheZoq2, 2020-06-18:
Would you mind wrapping your cargo.toml code in code blocks by surrounding it with ` ``` `? It's kind of hard to tell what's going on without it

TheZoq2, 2020-06-25:
Since this is more of a help thread than something that needs fixing/implementing, I'll go ahead and close it

Feel free to continue the discussion though :)

closed 2020-06-25 by TheZoq2