stm32-rs/stm32f1xx-hal #228: Serial::usartX(..)[.unwrap()]

pdgilbert, 2020-06-17:
Serial::usartX(..) in stm32f1xx-hal returns `-> Self` 
while stm32f4xx-hal and some others return `-> Result<Self, InvalidConfig>`
This of course means that using stm32f4xx-hal one must `unwrap()` or handle the error, while with stm32f1xx-hal one cannot `unwrap()`. 

It would be nice for portability, and to confuse newbies less, if the results were the same. Returning a `Result` structure seems more Rust-like, so I am filing the issue here. But I am a newbie so if there is a good reason not to return `Result` then I would appreciate an explanation and will file the issue with other HALs instead.

TheZoq2, 2020-06-18:
Hmm, I'm not sure I like the idea of introducing `Result` where nothing can go wrong as it might be an indication to a user that something *might* go wrong, and they need to handle the error.

Do you have a use case in mind where it is an advantage to have the same behaviour across HALs?  I could see the point in macros

> It would be nice for portability, and to confuse newbies less, if the results were the same.

I can't really speak for the confusion part, but I think it *might* lead to issues with portability if going from `f1` to `f4`.

If someone starts off with `f1`, they might write something like this

```rust
// NOTE: This is a safe unwrap, since the error returned is `Infallible`
let serial = uartX(...).unwrap();
```

At a later date, they decide to copy the same code over to the f4 crate and
assume that the unwrap is safe there too, which leads to an unhandled error.

If the return types are different, they will get a clear indication that that
behaviour isn't the same across crates

TeXitoi, 2020-06-18:
Even if I agree with your response, I really think that we need consistency in the stm32 hal APIs.

In this case, maybe that's even the philosophy of the api that is different: in one case it can't fail and try to do about what is asked, and the other does exactly what asked and fail if that's not possible. (I didn't look if that's really the case)

TheZoq2, 2020-06-18:
https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/serial.rs#L314

That does look like it is the case. Here is a panic which could be converted to an error

pdgilbert, 2020-06-18:
I was primarily thinking about consistency being the philosophy of the api.

 I think there are two portability aspects, one is moving the code and the other is developers shifting their thinking when changing hardware. My main real use case is that I have a project for which i have not yet committed to the hardware and I am experimenting so that I understand the relative merits of different hardware. A second more artificial case is that while learning Rust I have been trying to put together some examples that work with different stm32*-hal's and they tend to be littered with (unnecessary?) `#[cfg(feature = "stm32??xx")]` code. (See https://github.com/pdgilbert/eg_stm_hal if you want more details.) I am working on a list of some differences but this is the first one, other than `stm32::` vs `pac::`, that I've understood well enough to see that it is not forced by differences in the underlying hardware.

(BTW regarding "nothing can go wrong" my experience has been that something can almost always go wrong, and I usually bump into it. Since a possible panic has already been identified I think I may have filed this issue in the right place.)

pdgilbert, 2020-12-13:
I think this has been resolved.

closed 2020-12-13 by pdgilbert