stm32-rs/stm32f1xx-hal #238: BluePill pinout error (PB5 -> T3C2)

spam-receiver, 2020-06-25:
[Blue Pill pinout.pdf](https://github.com/stm32-rs/stm32f1xx-hal/files/4833837/Blue.Pill.pinout.pdf)

[image]

TheZoq2, 2020-06-27:
I'm not sure I understand what the issue is here. Are you saying that PB5 should be mapped to Timer 3 channel 2? If so, that is the case, but you need to use the `Tim3PartialRemap`, i.e. you then also have to remap channels 1-4 as (PB4, PB5, PB0, PB1)

spam-receiver, 2020-06-28:
I just make a correction in the pinout according to [manufacturer's datasheet](https://www.st.com/resource/en/datasheet/stm32f103c8.pdf).

burrbull, 2020-06-29:
Can you make pull request with change you want? I still don't understand what is wrong.

spam-receiver, 2020-06-29:
Simply change BluePillPinout.jpg to this correct variant:

[image]

burrbull, 2020-06-29:
You make us think that code is incorrect.

spam-receiver, 2020-06-29:
)))

PR #240 (2020-07-02): replace picture
Closes #238
closed 2020-07-02 by TheZoq2