stm32-rs/stm32f1xx-hal #248: set_high / set_low reversed?

pdgilbert, 2020-07-16:
Following a suggestion  by  @TheZoq2  in https://github.com/stm32-rs/stm32f1xx-hal/issues/234#issuecomment-647268069  I have been working on some examples that are intended to separate HAL/MPU setup so that the main application part of the code is clearly distinguished from the hardware initialization. In the blink example below I am finding that set_high turns the led off on my bluepill  and on with three other boards using stm32f3xx-hal, stm32f4xx-hal, and stm32l1xx-hal. I am a newbie so could be missing something simple, but this does seem peculiar.

```
//! Blink  onboard LED if one is available, or PC13 otherwise.
//! compare  blink3 example and stm32f1xx_hal example blinky.rs.
//! 
//! stm32f1xx below uses PC13  which is onboard green LED on Bluepill.
//! stm32f3xx below uses PE15  which is onboard green LD6 (West) LED on STM32F303 Discovery kit.
//! stm32f4xx below uses PA5   which is onboard green LD2  LED on STM32F411RET6 Nucleo-64 board,
//!                              but can be configures to PB13.
//! stm32l1xx below uses PB6   On some STM32L1.. Discovery boards there are onboard LD3 and LD4 LEDs on PB7 
//!                            and PB6 but mine are defective and so tested with off board LED on PB6.
//! 

#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
extern crate panic_semihosting;

#[cfg(not(debug_assertions))]
extern crate panic_halt;

use cortex_m_rt::entry;
use asm_delay::{ AsmDelay, bitrate, };

#[cfg(feature = "stm32f1xx")]  //  eg blue pill stm32f103
use stm32f1xx_hal::{prelude::*,   
                     pac::Peripherals,
		     gpio::{gpioc::PC13, Output, PushPull,}, 
		     };

#[cfg(feature = "stm32f1xx")] 
use embedded_hal::digital::v2::OutputPin;

#[cfg(feature = "stm32f3xx")]  //  eg Discovery-stm32f303
use stm32f3xx_hal::{prelude::*,
                    stm32::Peripherals, 
		    gpio::{gpioe::PE15, Output, PushPull,}, 
		    };

#[cfg(feature = "stm32f4xx")] // eg Nucleo-64  stm32f411
use stm32f4xx_hal::{prelude::*,   
                    pac::Peripherals, 
    	            delay::Delay,
		    gpio::{gpioa::PA5, Output, PushPull,}, 
		    };

#[cfg(feature = "stm32f4xx")]  
use embedded_hal::digital::v2::OutputPin;

#[cfg(feature = "stm32l1xx") ] // eg  Discovery STM32L100 and Heltec lora_node STM32L151CCU6
use stm32l1xx_hal::{prelude::*, 
                    stm32::Peripherals,
		    gpio::{gpiob::PB6, Output, PushPull,}, 
                    };

#[cfg(feature = "stm32l1xx") ] 
use embedded_hal::digital::v2::OutputPin;

#[entry]
fn main() -> ! {

    // setup() does all  hal/MCU specific setup and returns generic hal device for use in main code.
    
    // 1. Get device specific peripherals
    // 2. Take ownership of the raw rcc (Reset and Clock Control) device and convert to  HAL structs
    // 3. Configure gpio pin as a push-pull output. 

    #[cfg(feature = "stm32f1xx")]
    fn setup() -> (PC13<Output<PushPull>>, AsmDelay) {
       
       let dp        = Peripherals::take().unwrap();
       let mut rcc   = dp.RCC.constrain(); 
       let mut gpioc = dp.GPIOC.split(&mut rcc.apb2);

       // return tuple  (led, delay)
       (gpioc.pc13.into_push_pull_output(&mut gpioc.crh),            // led on pc13
        AsmDelay::new(bitrate::U32BitrateExt::mhz(16)) )             // delay
       };

    #[cfg(feature = "stm32f3xx")]
    fn setup() -> (PE15<Output<PushPull>>, AsmDelay) {

       let dp        = Peripherals::take().unwrap();
       let mut rcc   = dp.RCC.constrain();
       let mut gpioe = dp.GPIOE.split(&mut rcc.ahb);

       // return tuple  (led, delay)
       (gpioe.pe15.into_push_pull_output(&mut gpioe.moder, &mut gpioe.otyper),  // led on pe15
        AsmDelay::new(bitrate::U32BitrateExt::mhz(16)) )                        // delay
       };

    #[cfg(feature = "stm32f4xx")]
    fn setup() -> (PA5<Output<PushPull>>, AsmDelay) {

       let dp    = Peripherals::take().unwrap();
       let gpioa = dp.GPIOA.split();

       // return tuple  (led, delay)
       (gpioa.pa5.into_push_pull_output(),                         // led on pa5
        AsmDelay::new(bitrate::U32BitrateExt::mhz(32)) )           // delay
       };

    #[cfg(feature = "stm32l1xx")]
    fn setup() -> (PB6<Output<PushPull>>, AsmDelay) {

       let dp    = Peripherals::take().unwrap();
       let gpiob = dp.GPIOB.split();

       // return tuple  (led, delay)
       (gpiob.pb6.into_push_pull_output(),                        // led on pb6
        AsmDelay::new(bitrate::U32BitrateExt::mhz(4)) )           // delay
       };

    // End of hal/MCU specific setup. Following should be generic code.

    let (mut led, mut delay)  = setup();

    let on  : u32 = 1000;
    let off : u32 = 3000;

    // Wait for the timer to trigger an update and change the state of the LED
    loop {
        let _r = led.set_high();  // set_high turns bluepill off, Discovery & Nucleo-64 on
        delay.delay_ms(on);
        let _r = led.set_low();   // set_low turns bluepill  on,  Discovery & Nucleo-64 off
        delay.delay_ms(off);
    }
}

```

therealprof, 2020-07-16:
@pdgilbert GPIO pins can source or sink current. Depending on which way around the LED is connected it will light up if you pull the pin low (sink) or high (source).

thalesfragoso, 2020-07-16:
> pins can source or sink current. Depending on which way around the LED is connected it will light up if you pull the pin low (sink) or high (source).

That's the case for the led on the bluepill, it's wired as active-low. The led cathode is the one connected to the pin, while the anode is connected to Vcc.

therealprof, 2020-07-16:
You do know that "the" bluepill doesn't exist? There're at least a dozen variants of that design... 😅

pdgilbert, 2020-07-17:
Thanks for explaining the reason, I suspected it might be something like that. Yes, I am aware there are different variants of bluebill, but I don't have a sense of how extensive the difference are.

closed 2020-07-17 by pdgilbert

therealprof, 2020-07-17:
> Yes, I am aware there are different variants of bluebill, but I don't have a sense of how extensive the difference are.

Some do not even come with a STM32F103R8 MCU. 😅

pdgilbert, 2020-07-21:
(Just to complete this thread for anyone that might read it.)

The example in the original post does not achieve the objective of containing the hardware configuration in the `setup()` function and making the rest of the code generic, because of different boards wiring leds as sink or source. That is, `.set_low()` and `.set_high()` can have opposite effects on different boards. To remedy this I have added an LED trait with on and off methods defined in `setup()`, so `.on(`) and `.off()` can be used in the generic code. The  complete revised example is below. This example will change when embedded_hal 1.0.0 is released, at least in the aspect of `v2::OutputPin`. A more recent version may be available at https://github.com/pdgilbert/eg_stm_hal.

```
//! Blink  onboard LED if one is available, or PC13 otherwise.
//! compare  blink3 example and stm32f1xx_hal example blinky.rs.
//! 
//! stm32f1xx below uses PC13  which is onboard green LED on Bluepill.
//! stm32f3xx below uses PE15  which is onboard green LD6 (West) LED on STM32F303 Discovery kit.
//! stm32f4xx below uses PA5   which is onboard green LD2  LED on STM32F411RET6 Nucleo-64 board,
//!                              but can be configured to PB13.
//! stm32l1xx below uses PB6   On some STM32L1.. Discovery boards there are onboard LD3 and LD4 LEDs on PB7 
//!                            and PB6 but mine are defective and so tested with off board LED on PB6.
//! 

#![deny(unsafe_code)]
#![no_std]
#![no_main]

#[cfg(debug_assertions)]
extern crate panic_semihosting;

#[cfg(not(debug_assertions))]
extern crate panic_halt;

use cortex_m_rt::entry;
use asm_delay::{ AsmDelay, bitrate, };

#[cfg(feature = "stm32f1xx")]  //  eg blue pill stm32f103
use stm32f1xx_hal::{prelude::*,   
                     pac::Peripherals,
                     gpio::{gpioc::PC13, Output, PushPull,}, 
                     };

#[cfg(feature = "stm32f1xx")] 
use embedded_hal::digital::v2::OutputPin;

#[cfg(feature = "stm32f3xx")]  //  eg Discovery-stm32f303
use stm32f3xx_hal::{prelude::*,
                    stm32::Peripherals, 
                    gpio::{gpioe::PE15, Output, PushPull,}, 
                    };

#[cfg(feature = "stm32f4xx")] // eg Nucleo-64  stm32f411
use stm32f4xx_hal::{prelude::*,   
                    pac::Peripherals, 
                    gpio::{gpioa::PA5, Output, PushPull,}, 
                    };

#[cfg(feature = "stm32f4xx")]  
use embedded_hal::digital::v2::OutputPin;

#[cfg(feature = "stm32l1xx") ] // eg  Discovery STM32L100 and Heltec lora_node STM32L151CCU6
use stm32l1xx_hal::{prelude::*, 
                    stm32::Peripherals,
                    gpio::{gpiob::PB6, Output, PushPull,}, 
                    };

#[cfg(feature = "stm32l1xx") ] 
use embedded_hal::digital::v2::OutputPin;

pub trait LED {
   fn  on(&mut self)  -> () ;
   fn off(&mut self)  -> () ;
}

#[entry]
fn main() -> ! {

[... 72 lines cut ...]
           fn  off(&mut self)  -> () { self.set_low().unwrap() }
           };

       // return tuple  (led, delay)
       (gpiob.pb6.into_push_pull_output(),                        // led on pb6 with on/off
        AsmDelay::new(bitrate::U32BitrateExt::mhz(4)) )           // delay
       };

    // End of hal/MCU specific setup. Following should be generic code.

    let (mut led, mut delay)  = setup();

    let on  : u32 = 1000;
    let off : u32 = 3000;

    // Wait for the timer to trigger an update and change the state of the LED
    loop {
        let _r = led.on();  
        delay.delay_ms(on);
        let _r = led.off(); 
        delay.delay_ms(off);
    }
}

```