stm32-rs/stm32f1xx-hal #250: Question: Measure time delta between two events

faulesocke, 2020-07-22:
I have an input signal (rectangle) and want to measure it's frequency. So, intuitively I would write an EXTI ISR that starts an (upwards counting) timer (when not already started) and then stop the timer once the ISR triggers the second time. The ISR would be triggered on the rising edge. Afterwards I would read the timer value and calculate the signal frequency from that.

However, I haven't found a way to read a timers value. There is `CountDownTimer::micros_since`, however that only tells me how many µs passed since the last timer update and the docs say, I should not use it to measure time.

Is there anything I am currently missing? I can't imagine that I'm the first one trying to do something like this.

thalesfragoso, 2020-07-23:
If you know that your signal frequency is bigger than the timer frequency, then you can use `micros_since`, the thing with `micros_since` is that it counts microseconds passed since the last update event, the note is to make sure people don't mistake it with `micros()` functions that use a free running timer.

You can configure your EXTI ISR to disable the timer after receiving the second signal and also configure a timer interrupt, that way a timer interrupt will never happen if your signal frequency is bigger than the timer frequency. If it happens you know that you didn't get the second EXTI in time and you can disregard this measuring and start again.

faulesocke, 2020-07-23:
Ah, I thought the update event occours whenever the counter is increased/decreased. But it turns out that I was wrong and update is actually synonymous for overflow.

I will follow your described approach, thank you very much!

closed 2020-07-23 by faulesocke