stm32-rs/stm32f1xx-hal #252: DWT clock cycle count stuck unless trace is enabled

bschwind, 2020-07-29:
I tried using a `MonoTimer` to measure time between two events (in my case, some button pushes). I noticed the time-based code was never firing, and dug into the code until I discovered that [`DWT::get_cycle_count`](https://docs.rs/stm32f1xx-hal/0.6.1/stm32f1xx_hal/stm32/struct.DWT.html#method.get_cycle_count) always returns `1073741824` (2^30)

In order for clock cycles to be successfully reported from `DWT`, I have to call `cp.DCB.enable_trace();`

Here is a minimal reproduction of the issue, printing the clock cycle counts over a serial interface. My hardware is a "black pill" `stm32f103c8t6`, specifically [this dev board](https://stm32-base.org/boards/STM32F103C8T6-Black-Pill). I flash and monitor output via a CP2102 serial-usb adapter. I'm _not_ currently using a debugger probe of any sort.

```rust
#![deny(unsafe_code)]
#![no_main]
#![no_std]

use panic_halt as _;

use cortex_m_rt::entry;
use stm32f1xx_hal::{
    pac,
    prelude::*,
    serial::{Config, Serial},
};

use core::fmt::Write;

#[entry]
fn main() -> ! {
    let mut cp = cortex_m::peripheral::Peripherals::take().unwrap();
    let p = pac::Peripherals::take().unwrap();

    let mut flash = p.FLASH.constrain();
    let mut rcc = p.RCC.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let mut afio = p.AFIO.constrain(&mut rcc.apb2);
    let mut gpioa = p.GPIOA.split(&mut rcc.apb2);

    let tx = gpioa.pa9.into_alternate_push_pull(&mut gpioa.crh);
    let rx = gpioa.pa10;

    let serial = Serial::usart1(
        p.USART1,
        (tx, rx),
        &mut afio.mapr,
        Config::default().baudrate(115200.bps()),
        clocks,
        &mut rcc.apb2,
    );

    let (mut tx, _rx) = serial.split();

    // Needed in order for MonoTimer to work properly
    cp.DCB.enable_trace();

    let mut delay = stm32f1xx_hal::delay::Delay::new(cp.SYST, clocks);
    let mono_timer = stm32f1xx_hal::time::MonoTimer::new(cp.DWT, clocks);

    let start = mono_timer.now();

    loop {
        let elapsed = start.elapsed();
        writeln!(tx, "Elapsed clock cycles {}\r", elapsed).unwrap();
        delay.delay_ms(100_u32);
    }
}
```

I found a few older issues in earlier repositories about this:

* https://github.com/japaric/stm32f103xx-hal/issues/76#issuecomment-417891794
* https://github.com/japaric/stm32f103xx-hal/pull/94

TheZoq2, 2020-08-01:
Huh, I didn't know you could enable the DWT without a debugger attached, we probably should do that then... Though it still feels a bit odd to use the debug peripheral for timekeeping

fuchsnj, 2020-08-07:
I just tested this on a bluepill, and _without_ `cp.DCB.enable_trace()`, `DWT::get_cycle_count()` always returns 1073741824 (same as @bschwind ). However, when I enable trace, it always returns 0.
I haven't been able to get it to actually count cycles yet.

fuchsnj, 2020-08-07:
Ah, spoke too soon. If I enable trace _before_ I create the `MonoTimer` everything is working fine on my bluepill. Enabling it after makes it always return 0.

PR #254 (2020-08-12): Fix mono timer
Closes #252 

Unfortunately this a breaking change as we now need to take the DCP peripheral when starting the mono timer.

I also added a note from one of the linked issues about `mono_timer` being stopped when hprintln is running.

For reference, here is a code sample that does not work currently, but does with this fix

```rust
#![deny(unsafe_code)]
#![no_main]
#![no_std]

use panic_halt as _;

use cortex_m_semihosting::hprintln;
use cortex_m_rt::entry;
use stm32f1xx_hal::{
    pac,
    prelude::*,
    serial::{Config, Serial},
};
use embedded_hal::digital::v2::OutputPin;

us [... cut]
closed 2020-08-12 by therealprof