stm32-rs/stm32f1xx-hal #259: Firmware panics with f32 arithmetic, but not with f64

bschwind, 2020-08-19:
I first discovered this issue when trying to set a new duty value for PWM. I just wanted to multiply the max duty by some fractional float, like `0.8`. I eventually noticed my firmware was halting when I reached a multiplication instruction with an `f32` value. Forgive me if this isn't the right place to raise this issue, but I've never seen an issue like this before and it felt like the most appropriate place to file it.

I'm using an [stm32f103 board](https://stm32-base.org/boards/STM32F103C8T6-Black-Pill) via a UART connection. The target is `thumbv7em-none-eabi`, which notably doesn't have a floating point unit. I would think the code works in either case, just slower without a floating point unit.

## Sample Code

```rust
#![no_main]
#![no_std]

use panic_halt as _; // panic handler

use stm32f1xx_hal as hal;

use core::fmt::Write;
use cortex_m_rt::entry;
use hal::{
    pac,
    prelude::*,
    serial::Config as SerialConfig,
    timer::{Timer},
};

#[entry]
fn main() -> ! {
    let cp = cortex_m::peripheral::Peripherals::take().expect("failed to get cortex_m peripherals");
    let dp = pac::Peripherals::take().expect("failed to get stm32 peripherals");

    // Take ownership over the raw flash and rcc devices and convert them into the corresponding
    // HAL structs.
    // RCC = Reset and Clock Control
    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    // Prepare the alternate function I/O registers
    let mut afio = dp.AFIO.constrain(&mut rcc.apb2);

    // Grab the GPIO banks we'll use.
    let mut gpioa = dp.GPIOA.split(&mut rcc.apb2);

    // Create a delay abstraction based on SysTick.
    let mut delay = hal::delay::Delay::new(cp.SYST, clocks);

    // Set up serial communication on pins A9 (Tx) and A10 (Rx), with 115200 baud.
    let serial_config = SerialConfig::default().baudrate(115200.bps());
    let usart1 = dp.USART1;
    let usart_pins = (gpioa.pa9.into_alternate_push_pull(&mut gpioa.crh), gpioa.pa10);
    let mapr = &mut afio.mapr;
    let apb = &mut rcc.apb2;
    let usart = hal::serial::Serial::usart1(usart1, usart_pins, mapr, serial_config, clocks, apb);
    let (mut tx, _rx) = usart.split();

    // PWM Setup
    let pwm_pins = gpioa.pa8.into_alternate_push_pull(&mut gpioa.crh);
    let mut pwm =
        Timer::tim1(dp.TIM1, &clocks, &mut rcc.apb2).pwm(pwm_pins, &mut afio.mapr, 1.khz()).split();

    pwm.enable();

    writeln!(
        tx,
        "Before floating point\r",
    )
    .unwrap();

    // The commented line below doesn't work, but the uncommented one does.
    // let new_duty = (pwm.get_max_duty() as f32 * 0.95) as u16;
    let new_duty = (pwm.get_max_duty() as f64 * 0.95) as u16;
    writeln!(
        tx,
        "New duty: {}\r",
        new_duty
    )
    .unwrap();

    pwm.set_duty(new_duty);
    
    writeln!(
        tx,
        "After floating point\r",
    )
    .unwrap();

    loop {
        delay.delay_ms(10_u32);
    }
}

```

## Output (`f32`):

```
Before floating point
```

## Output (`f64`)

```
Before floating point
New duty: 7600
After floating point
```

Is there something fundamental I'm missing about floating point numbers and conversions into other primitive types?

TheZoq2, 2020-08-19:
Wow, that's a strange issue. My first thought was that you have an incorrect target specified, but that seems like it's not the case.

Are you able to reproduce this with a more minimal example? For example: are you able to get rid of all of the program apart from the float computation? Can you do other similar float math?

Are you able to run a debugger to see exactly where it fails, and where it gets stuck once it fails?

bschwind, 2020-08-19:
I don't have a debugger but I think I can turn another stm32 into one? I'll try that tomorrow and come back with the results.

TheZoq2, 2020-08-19:
Not something I've done, but maybe..

TeXitoi, 2020-08-19:
> I don't have a debugger but I think I can turn another stm32 into one? I'll try that tomorrow and come back with the results.

https://satoshinm.github.io/blog/171223_jtagswdpillblink_jtagswd_debugging_via_black_magic_probe_on_an_stm32_blue_pill_and_blinking_a_led_using_stm32cubemx_libopencm3_and_bare_metal_c.html

Didn't tested it.

bschwind, 2020-08-24:
Here's what I have so far from `gdb` attached via the black magic probe:

```
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: .../target/thumbv7em-none-eabi/release/stm32-test 

Program received signal SIGSEGV, Segmentation fault.
compiler_builtins::float::mul::mul::h0a665951c2146d55 () at /cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.31/src/float/mul.rs:43
43	/cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.31/src/float/mul.rs: No such file or directory.
(gdb) bt
#0  compiler_builtins::float::mul::mul::h0a665951c2146d55 () at /cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.31/src/float/mul.rs:43
#1  compiler_builtins::float::mul::__mulsf3::h0de126c916758246 () at /cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.31/src/float/mul.rs:183
#2  0x0800039c in stm32_test::__cortex_m_rt_main::h0b5a596fbc26c020 ()
#3  0x0800013a in main ()
```

thalesfragoso, 2020-08-24:
@bschwind Not sure if that is the cause, but you should be using thumbv7m-none-eabi for the M3, not thumbv7**e**m-none-eabi.

bschwind, 2020-08-25:
@thalesfragoso good catch, however now I get a different backtrace:

```
Starting program: .../target/thumbv7m-none-eabi/release/stm32-test 

Program received signal SIGSEGV, Segmentation fault.
0x08002834 in ?? ()
(gdb) bt
#0  0x08002834 in ?? ()
#1  0x0800039c in stm32_test::__cortex_m_rt_main::he96b498b81305024 ()
#2  0x0800039c in stm32_test::__cortex_m_rt_main::he96b498b81305024 ()
#3  0x0800013a in main ()
```

I'll have to try making a more minimal example which doesn't optimize the float calculations in the compile step.

bschwind, 2020-08-25:
Ah, I may have spoke too soon. I removed a bit of the serial code and got it running. I'll experiment a bit more but I think changing the target may have fixed main reason I opened the issue, thanks for pointing that out!

TheZoq2, 2020-08-29:
You're not the first to fall in that trap :) I even checked your target when I first read the issue but also missed the `m` vs `em` thing :D 

I'll close this for now, but feel free to re-open or continue the discussion if something else arrises

closed 2020-08-29 by TheZoq2